rtl-airband/RTLSDR-Airband #1: Compiling < gcc 4.8

rec9140, 2015-04-27:
*** THIS IS HERE SO WE CAN CONTINUE on the NEW DESKTOP VERSION v. the older SoC limited one
*** RESOLUTION to this in diff format BELOW!!! ***

$ sudo apt-get install libmp3lame-dev libvorbis-dev libshout-dev libfftw3-dev
 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libshout3-dev' instead of 'libshout-dev'
libfftw3-dev is already the newest version.

The following extra packages will be installed:
  libogg-dev libspeex-dev libtheora-dev
The following NEW packages will be installed:
  libmp3lame-dev libogg-dev libshout3-dev libspeex-dev libtheora-dev libvorbis-dev
0 upgraded, 6 newly installed, 0 to remove and 50 not upgraded.
Need to get 1,197 kB of archives.
After this operation, 4,829 kB of additional disk space will be used.
Do you want to continue [Y/n]? y
Get:1 http://us.archive.ubuntu.com/ubuntu/ precise/universe libmp3lame-dev i386 3.99.3+repack1-1 [184 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu/ precise/main libogg-dev i386 1.2.2~dfsg-1ubuntu1 [63.3 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu/ precise/main libspeex-dev i386 1.2~rc1-3ubuntu2 [69.0 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu/ precise/main libtheora-dev i386 1.1.1+dfsg.1-3ubuntu2 [393 kB]
Get:5 http://us.archive.ubuntu.com/ubuntu/ precise/main libvorbis-dev i386 1.3.2-1ubuntu3 [448 kB]
Get:6 http://us.archive.ubuntu.com/ubuntu/ precise/main libshout3-dev i386 2.2.2-7ubuntu1 [39.1 kB]
Fetched 1,197 kB in 2s (513 kB/s)        
Selecting previously unselected package libmp3lame-dev.
(Reading database ... 442151 files and directories currently installed.)
Unpacking libmp3lame-dev (from .../libmp3lame-dev_3.99.3+repack1-1_i386.deb) ...
Selecting previously unselected package libogg-dev.
Unpacking libogg-dev (from .../libogg-dev_1.2.2~dfsg-1ubuntu1_i386.deb) ...
Selecting previously unselected package libspeex-dev.
Unpacking libspeex-dev (from .../libspeex-dev_1.2~rc1-3ubuntu2_i386.deb) ...
Selecting previously unselected package libtheora-dev.
Unpacking libtheora-dev (from .../libtheora-dev_1.1.1+dfsg.1-3ubuntu2_i386.deb) ...
Selecting previously unselected package libvorbis-dev.
Unpacking libvorbis-dev (from .../libvorbis-dev_1.3.2-1ubuntu3_i386.deb) ...
Selecting previously unselected package libshout3-dev.
Unpacking libshout3-dev (from .../libshout3-dev_2.2.2-7ubuntu1_i386.deb) ...
Setting up libmp3lame-dev (3.99.3+repack1-1) ...
Setting up libogg-dev (1.2.2~dfsg-1ubuntu1) ...
Setting up libspeex-dev (1.2~rc1-3ubuntu2) ...
Setting up libtheora-dev (1.1.1+dfsg.1-3ubuntu2) ...
Setting up libvorbis-dev (1.3.2-1ubuntu3) ...
Setting up libshout3-dev (2.2.2-7ubuntu1) ...
~/RTLSDR-Airband$ make rtl_airband
g++ -O3 -march=native    rtl_airband.cpp  -lrt -lm -lvorbisenc -lmp3lame -lshout -lpthread -lrtlsdr -lfftw3f -o rtl_airband
rtl_airband.cpp: In function ‘int main(int, char**)’:
rtl_airband.cpp:600:33: error: ‘__builtin_cpu_supports’ was not declared in this scope
rtl_airband.cpp:624:37: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
rtl_airband.cpp:667:121: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
rtl_airband.cpp:685:172: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
make: *** [rtl_airband] Error 1

*** NOTE: the resolution is:

$ diff rtl_airband.cpp rtl_airband.cpp.bak.orig 
600c600
<       if(1) {
---
>       if(__builtin_cpu_supports("sse")) {

This results in the build of rtl_airband in the directory.

And it does seem to run.... I will have to connect up some additional RTL's and get some antenna work done to test things more. My main RTL is in use on mission critical op right now....

So the instructions say to run ./rtl_airband_neon

BUT 

this was not built... just rtl_airband

Is the neon version just for the SoC devices?

szpajder, 2015-04-27:
Yes, _neon is for Raspberry Pi V2, while _vfp is for Pi V1. x86 binary has
no suffix.

rec9140, 2015-04-27:
ok.. might want to update the x86 build info ;) 

So I have it built, and dug out a 2nd device, and running... no antenna yet... as I have to get setup for multiple dongles...

Getting feeds to the server, and what I guess is the "scanning" screen... So looks GOOD! I will advise as soon as I can wire in my antenna(s) and compare with my regular receivers... but looks good!

THAAANKKKKKKKSSSSS!

I've got a couple other comments for issues/suggestions for the future....

but

THANKKKKKKKKKSSSSSSSSSSSSS AGAIN!!!!! DESKTOP BUILD WOOOT WOOOWOOT!!!

szpajder, 2015-04-28:
On Mon, Apr 27, 2015 at 11:07 PM, rec9140 <notifications@github.com> wrote:

> ok.. might want to update the x86 build info ;)
>

It's now updated. Thanks for pointing this out.

-- 
Tomasz Lemiech

commit 00c205ab (2015-05-07): Implement SSE detection in a more portable way

gcc versions older than 4.8 are still common in the field and
they do not support __builtin_cpu_supports() function. Switched
to old-style inline asm macro for some time.

Closes #1.
closed 2015-05-07 by szpajder