rtl-airband/RTLSDR-Airband #3: [Suggesiton] Noise reduction code

thetobby, 2015-05-30:
Perhaps this codebase can be useful?
http://habrahabr.ru/post/243977/

Sourcecode:
http://cgit.freedesktop.org/pulseaudio/webrtc-audio-processing/

thetobby, 2015-05-30:
An easier implementation may be to just route the audio through a loopback audio interface with pulseaudio since the audio is not time sensitive instead of adding the code into the program itself.

bovine, 2016-06-23:
@thetobby are you saying that you've tested the effect of the "echo-cancellation" filter and found that to be the most effective one? It looks like there is also "noise-supression" filter in that codebase which sounds like it might be the better option:

https://cgit.freedesktop.org/pulseaudio/webrtc-audio-processing/tree/webrtc/modules/audio_processing

barjac, 2018-01-03:
I find that most 'noise' that impairs legibility of air-band reception is above the required bandwidth for speech and can be removed by using the graphic equalizer and low pass filter in PulseEffects, set to cut around 2.7kHz.
https://github.com/wwmm/pulseeffects

cblknittights, 2018-01-03:
>can be removed by using the graphic equalizer and low pass filter in PulseEffects

Not everyone uses pulse* . I am strictly ALSA, and pulse* is one of the first things I apt-get purge on any setup.

Strykar, 2019-03-14:
Any noise reduction code would be useful, even using Sox.
I'm not a fan of Pulse tho I do use it on some desktops.

danmilon, 2019-03-14:
https://github.com/szpajder/RTLSDR-Airband/pull/112 is related.

charlie-foxtrot, 2023-04-21:
no additional plans to add noise reduction code, closing

closed 2023-04-21 by charlie-foxtrot