rtl-airband/RTLSDR-Airband #5: USB dropping samples

thetobby, 2015-06-08:
My USB driver/hw seems to be dropping packets when reproducing the hardcoded settings with rtl_test -s 2.5e6 on my desktop.
To ensure better comparability with different hardware, may I suggest adding custom bandwidth in the configuration or reduce to 2.4?

If you want I can provide some numbers from R820T, R820T2 and E4000 on raspberry pi v2 to help determine the limitation with that perticular hardware.

szpajder, 2015-06-09:
Yes, that's feasible - however I have to rearrange things a bit first,
because there are several buffers used in the code and their size depend on
each other and on SOURCE_RATE as well, so it's not as simple as changing a
single #define . Stay tuned.

thetobby, 2015-06-13:
Just found another need for this. I'm upgrading to 4 dongles today, potentially 6 to get the remaining two dongles to cover one UHF frequency each that is spaced faar apart, but since It pulls the full bandwidth when I only need one channel  I'm afraid it might hit the hw roof.

szpajder, 2018-01-03:
The current code in unstable branch implements a `sample_rate` option (in the device section) which does exactly this.

szpajder, 2018-02-10:
Support for configurable sampling rates has been released in version 3.0.0.

Closing.

closed 2018-02-10 by szpajder