rtl-airband/RTLSDR-Airband #6: Possible to run on cubieboard?

snip, 2015-06-27:
Do you think it is possible to run it on Cubieboard?

http://docs.cubieboard.org/products/start
---
$ cat /proc/cpuinfo 
Processor       : ARMv7 Processor rev 4 (v7l)
processor       : 0
BogoMIPS        : 668.05

processor       : 1
BogoMIPS        : 670.35

Features        : swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 4

Hardware        : sun7i
Revision        : 0000
Serial          : 0000000000000000
---

$ gcc -c -Q --help=target
The following options are target specific:
  -mabi=                        
  -mabort-on-noreturn                   [disabled]
  -mandroid                             [disabled]
  -mapcs                                [disabled]
  -mapcs-float                          [disabled]
  -mapcs-frame                          [disabled]
  -mapcs-reentrant                      [disabled]
  -mapcs-stack-check                    [disabled]
  -march=                               armv7-a
  -marm                                 [disabled]
  -mbig-endian                          [disabled]
  -mbionic                              [disabled]
  -mcallee-super-interworking           [disabled]
  -mcaller-super-interworking           [disabled]
  -mcirrus-fix-invalid-insns            [disabled]
  -mcpu=                        
  -mfix-cortex-m3-ldrd                  [enabled]
  -mfloat-abi=                          hard
  -mfp16-format=                
  -mfp=                         
  -mfpe                                 [disabled]
  -mfpe=                        
  -mfpu=                                vfpv3-d16
  -mglibc                               [enabled]
  -mhard-float                          [disabled]
  -mlittle-endian                       [enabled]
  -mlong-calls                          [disabled]
  -mpic-register=               
  -mpoke-function-name                  [disabled]
  -msched-prolog                        [enabled]
  -msingle-pic-base                     [disabled]
  -msoft-float                          [disabled]
  -mstructure-size-boundary=    
  -mthumb                               [enabled]
  -mthumb-interwork                     [enabled]
  -mtp=                         
  -mtpcs-frame                          [disabled]
  -mtpcs-leaf-frame                     [disabled]
  -mtune=                       
  -muclibc                              [disabled]
  -mvectorize-with-neon-quad            [disabled]
  -mword-relocations                    [disabled]
  -mwords-little-endian                 [disabled]
---
Thanks

szpajder, 2015-06-27:
Right now it would not work because the FFT code is specific to Broadcom GPU, while Cubieboard uses Mali400. The quick and dirty way is to get rid of hello_fft and calculate FFT on the main CPU (eg. with fftw3 library). However it would probably be too CPU-intensive. The right way is to do this on Mali, however I currently don't have such hardware at hand to experiment.

snip, 2015-06-27:
Yes i was thinking about a no GPU processing. Is it a hard work to do it?

szpajder, 2015-06-27:
Here is a quick and dirty FFT benchmark using FFTW3 library:

        #include <stdio.h>
        #include <fftw3.h>
        #include <sys/time.h>
        #define FFT_SIZE 512
        #define CNT 2560000*10
        int main() {
            fftwf_plan fft;
            fftwf_complex* fftin;
            fftwf_complex* fftout;
            fftin = fftwf_alloc_complex(FFT_SIZE);
            fftout = fftwf_alloc_complex(FFT_SIZE);
            fft = fftwf_plan_dft_1d(FFT_SIZE, fftin, fftout, FFTW_FORWARD, FFTW_MEASURE);
            int i,j;
            for(j=0; j<CNT/FFT_SIZE; j++) {
                for(i=0; i<FFT_SIZE; i++) {
                    if(i % 4 == 0)
                        fftin[i][0]=10.0;
                    else
                        fftin[i][0]=0.0;
                    fftin[i][1]=0.0;
                }
                fftwf_execute(fft);
            }
        }

Save it as fft.c compile with: `gcc -lfftw3f -o fft fft.c` and run with `time ./fft`. You will see how long it takes to calculate FFT on the CPU for 10 seconds worth of rtl-sdr samples. On my RPiv2 it takes a bit over 4 seconds. So we're going to hit 40% CPU utilization only on doing FFTs and there is still work left to do (signal windowing, demodulation, AGC, auto-squelch, MP3 compression, streaming to Icecast). I believe it would be barely usable even with a single dongle. IMHO going for FFT on CPU is not worth the effort.

snip, 2015-06-27:
On a Cubieboard2:
'''
$ time ./fft

real    0m2.056s
user    0m2.040s
sys     0m0.010s
'''

szpajder, 2015-06-27:
Alright then, please pull this branch: https://github.com/szpajder/RTLSDR-Airband/tree/arm7v_nogpu

Compile with `make rtl_airband` (not _neon, not _vfp).

It's just a quick hack for you to test the performance, so it might be broken. I can't use it on RPiv1 - it gives me 100% CPU and the stream stutters horribly.

snip, 2015-06-27:
Thanks!

Running and using 48% of CPU without anyone speaking.

I will do some more tests by tomorrow with some activity.

Thanks again.

snip, 2015-06-28:
52% of CPU when listening to 3 frequencies (on single dongle) with one with activity.
So it works very well!

This seems to be a good solution for generic CPU.

szpajder, 2015-06-29:
Thanks for the test. I'll then make it available as a compile time option, but it will take a while, because I have to restructure the code a bit first (all this ifdef stuff is a huge mess right now).

For the timebeing, just use the hack from the arm7v_nogpu branch.

snip, 2015-06-29:
Ok no problem. Thanks for the accomplished work.
And we are awaiting your unstable branch release  with very cool features.

szpajder, 2015-12-26:
This issue has been addressed by https://github.com/szpajder/RTLSDR-Airband/commit/3171289e7f52f2a41e5c7c59ef619edbf7d536ba

Closing.

commit 3171289e (2015-12-26): armv7-generic build target

It is now possible to run the program on platforms like Cubieboard,
which use ARMv7 CPU, but do not have Broadcom VideoCore GPU. Main
CPU is therefore used for FFT calculation, so performance impact is
significant. To use the new target, build the program with:

PLATFORM=armv7-generic make

Closes #6.
closed 2015-12-26 by szpajder