rtl-airband/RTLSDR-Airband #7: Hostname, username, password...

OZ1SEJ, 2015-07-23:
I've installed it on my raspberry pi, but I'm stuck in config.txt. What hostname, username and password should I fill in here?

szpajder, 2015-07-23:
rtl_airband does not output anything to the local soundcard. You need an Icecast server to stream the audio to.

You can set up your own server using Icecast software from http://icecast.org/ . It is bundled in Raspbian (package named icecast2), but you can use an external one as well.

Icecast requires you to configure source authentication (username and password). Only authenticated sources cand send streams to the server. You put these settings in icecast.xml configuration file.

Then you need to tell rtl_airband where is your Icecast server (hostname, port) how to authenticate to it (username, password) and the mount point path. For example, if your Icecast server is "example.com", port 8000 and the mount point path is "mystream", then in order to listen to the stream you should point your client (VLC, Winamp, Windows Media Player, whatever) to http://example.com:8000/mystream.

OZ1SEJ, 2015-07-24:
Okay, so now, I've installed Icecast2. But I still need some information to make it run.

I've followed the instructions, so now, I have a directory called /home/pi/rtl-sdr/ containing the rtl-sdr program, which works. I can listen to radio with SDR♯ on my laptop.

In this directory, I have a "build" directory, and inside that, I have an RTLSDR-Airband directory.

Contents of the file /home/pi/rtl-sdr/build/RTLSDR-Airband/config.txt:

    1
    0       3       280     119325000       -8000
    localhost       8000    118450.mp3      118450000       source  asdf1234
    localhost       8000    119800.mp3      119800000       source  asdf1234
    localhost       8000    120200.mp3      120200000       source  asdf1234

If 280 is the gain in dB, that's mighty high. I usually run it around 32 dB.

If -8000 is the frequency correction in ppm, that's a lot. I usually only need +37 ppm.

I don't have any files by the name 118450.mp3 anywhere, but maybe I don't need to?

I also have a file /home/pi/rtl-sdr/build/RTLSDR-Airband/config/template.txt. Should I edit this file? Should it still be called template.txt? Should I keep the angle brackets in the file?

I also have a file called /etc/icecast2/icecast.xml, but I don't know how to configure it. Could you provide an example?

OZ1SEJ, 2015-07-25:
Okay, I think I've everything set up now, including Icecast. But when I run ./rtl_airband_neon, I get "Illegal instruction". When I run "sudo ./rtl_airband_neon" I just get nothing. The program runs and stops, and I'm back at the prompt. By the way, my config/template.txt file now looks like this:

    <localhost>
    <8000>
    <source>
    <asdf1234>
    <119325000>
    <-8000>
    <280>
    <118450000> <118450.mp3>
    <119800000> <119800.mp3>
    <120200000> <120200.mp3>

szpajder, 2015-07-26:
On Fri, Jul 24, 2015 at 9:32 PM, Steen Eiler Jørgensen <
notifications@github.com> wrote:
>
> If 280 is the gain in dB, that's mighty high. I usually run it around 32
> dB.
>
It's the gain in tenths of dB. If you want 32 dB, put 320 here.

> If -8000 is the frequency correction in ppm, that's a lot. I usually only
> need +37 ppm.
>
Nope, it's not in ppm, it's in hertz, negative. If you want +37ppm, put
-3700 here.

I don't have any files by the name 118450.mp3 anywhere, but maybe I don't
> need to?
>

These are not files, but mountpoint names on the Icecast server. You don't
need to create them in advance. Icecast server creates the mountpoint
automatically as soon as somebody starts feeding to it.

> I also have a file
> /home/pi/rtl-sdr/build/RTLSDR-Airband/config/template.txt. Should I edit
> this file? Should it still be called template.txt? Should I keep the angle
> brackets in the file?
>

It's obsolete (actually it's no longer there in unstable branch and will be
removed from master branch soon). Only `config.txt` is used for
configuration.

> I also have a file called /etc/icecast2/icecast.xml, but I don't know how
> to configure it. Could you provide an example?
>
See here:
https://github.com/microtony/RTLSDR-Airband/issues/12#issuecomment-117054460

But when I run ./rtl_airband_neon, I get "Illegal instruction".
>

Most probably you have Raspberry Pi V1, right? Then you should `make
rtl_airband_vfp`. rtl_airband_neon is for RPi V2.

-- 
Tomasz Lemiech

OZ1SEJ, 2015-07-28:
Ah! I had a RPi V1 - I thought it was a V2 :-) Everything works now - thank you so much! :)

OZ1SEJ, 2015-07-28:
Oh - one extra question: If I make changes to config.txt (e.g. gain), will they be implemented immediately, or do I need to restart the application?

szpajder, 2015-07-28:
You need to restart.

hamav8tor, 2015-07-28:
You will need to restart.

OZ1SEJ, 2015-07-28:
Great - thanks!

closed 2015-12-26 by szpajder