rtl-airband/RTLSDR-Airband #11: Can't open device file: char_dev

Xam1987, 2015-12-11:
Hi,

I am trying to run RTLSDR-Airband on my rpi. I clone this repository and compiled the code successfully. 

Unfortunately, I get a "Can't open device file: char_dev" message when I try to run the program. I **did** followed all instructions  - created the device file with "sudo mknod char_dev c 100 0" and ran the program with sudo.

Running the program with strace indicated that there is a problem opening the char_dev special file, but I could not understand why it failed.
open("char_dev", O_RDONLY)              = -1 ENXIO (No such device or address)

here is ls -la output on char_dev. Did I created it right?

crw-r--r--  1 root root 100, 0 Dec 11 17:17 char_dev

Thanks

szpajder, 2015-12-11:
Since Linux kernel version 4.1 the VideoCore driver uses major number 249
instead of 100. So either please recreate char_dev with major 249 or just
put a symlink to /dev/vcio, which is created by default in recent Raspbian
trees.

The code in master branch is a bit outdated, I recommend using unstable
branch. It still needs a bit of polishing before merging back into master,
however it's already pretty stable and has some cool new features. And it
uses /dev/vcio by default, so no need to create char_dev in the working dir.

Xam1987, 2015-12-11:
Thanks for this quick response! I just looked through hello_fft sample code and figured out that /dev/vcio is what i'm looking for. 

I'll try the unstable branch.

closed 2015-12-11 by Xam1987