rtl-airband/RTLSDR-Airband #14: raspistill hangs

sutregutt, 2016-01-01:
Hi!
I have Raspberry camera module installed on my pi2. Do you have any idea why raspistill program hangs when /usr/local/bin/rtl_airband is running? If I kill the rtl_airband-process, raspistill runs well.
(same with v. 1 and v. 2)

Regards, 
Vegard

szpajder, 2016-01-01:
It probably needs exclusive access to the graphics chip. rtl_airband also
uses this to speed up FFT calculations. Only one program can access
/dev/vcio at the same time.

On RPiV2 you may try to compile rtl_airband with PLATFORM=armv7_generic.
This will omit the GPU code - FFTs will be calculated using main CPU, so
expect CPU utilization at 60% or more.

jaimehempke, 2016-01-19:
Good evening, I have used with great success the previous version without any problems, but with the new version can not keep it running for more than a few minutes, then she stops working, just get a new transmission or attempt to record more than one channel, already tried everything and I can not run, the most important to me in this new version is the possibility of recording.
My Settings are as follows: 2 dongles ubuntu linux 14, 8 channels dongle, also tested on other computers.
usually the error code is: Jan 17 10:08:25 Fonia kernel: [202.535141] rtl_airband [1831] segfault at 47 ip b70c7717 sp b3144280 error 4 in libc-2.21.so [b7067000 1b4000 +]
You have any solutions?
Thank you for software quality.

closed 2016-04-03 by szpajder