rtl-airband/RTLSDR-Airband #15: Erro libc

jaimehempke, 2016-01-19:
Jan 17 10:08:25 Fonia kernel: [  202.535141] rtl_airband[1831]: segfault at 47 ip b70c7717 sp b3144280 error 4 in libc-2.21.so[b7067000+1b4000]
Good morning , I am unable to run the new version for more than a few minutes , I always have this error above , already tried everything , I use ubuntu linux 14 . Also use two dongles , the function that interested me most was the recording option . The old version have no problems , what can I do?

szpajder, 2016-01-19:
What hardware platform are you running on?

Can you please post a gdb backtrace, when it crashes?

Example (run as root):

    gdb /path/to/rtl_airband

From gdb prompt run this command:

    run -f

When it crashes, you should see a GDB message "Program received signal SIGSEGV, Segmentation fault". Then run this command:

    thread apply all bt

and post the output.

Thanks in advance.

jaimehempke, 2016-01-20:
I have some difficulty in understanding due to language, I am in Brazil, I try to explain step by step the command ok.
I did the following gdb / usr / local / bin / rtl_airband -f returned me the following message:

jaime@jaime-TA880GB:~$ gdb  /usr/local/bin/rtl_airband 
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Lendo símbolos de /usr/local/bin/rtl_airband...concluído.
(gdb)

jaimehempke, 2016-01-20:
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb4244b40 (LWP 2939)]
_IO_new_fclose (fp=0x61657669) at iofclose.c:  7iofclose.c: Arquivo ou diretÃ³rio nÃ£o encontrado.
(gdb) thread apply all bt2      
Thread 7 (Thread 0xb4244b40 (LWP 2939)):
#0  _IO_new_fclose (fp=0x61657669) at iofclose.c:49 6   
#1  0x0804d347 in process_outputs (channel=channel@entry=0xb6dbb470)
    at rtl_airband.cpp:431
#2  0x0804d6cd in output_thread (params=0x0) at rtl_airband.cpp:475------------
#3  0xb7f1af70 in start_thread (arg=0xb4244b40) at pthread_create.c:312
#4  0xb7ae9bee in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:129

Thread 6 (Thread 0xb4a45b40 (LWP 2938)):
#0  0xb7fdccb0 in ?? ()
#1  0x56974e81 in ?? ()
#2  0x73b060a3 in ?? ()
#3  0x9ba687b1 in ?? ()
#4  0xb186a79a in ?? ()
#5  0xa361b072 in ?? ()
#6  0x814d9755 in ?? ()
#7  0x5d5f6f51 in ?? ()
#8  0x4d82536c in ?? ()
#9  0x62a65296 in ?? ()
#10 0x88b171ad in ?? ()
#11 0xa8989aaa in ?? ()
#12 0xaf72ae88 in ?? ()
---Type <return> to continue, or q <return> to quit---

szpajder, 2016-01-20:
I have pushed a fix to the hotfix branch.

Please do the following:
 * go to https://github.com/szpajder/RTLSDR-Airband/tree/hotfix
 * download the tree (use "Download ZIP" button), unpack it into a new directory
 * compile it as usual
 * test it and let me know if it's OK now.

Thanks.

jaimehempke, 2016-01-21:
Unfortunately it did not work, did as instructed but as soon as you receive a signal more appears the same erro.Também does not work when I activate more than one frequency in the second donlge ja software segmentação.estou failure sending the two errors to you the first is what was already appearing and the second is the error of plus frequencies in the second dongle

Thread 7 (Thread 0xb4231b40 (LWP 2954)):
#0  _IO_new_fclose (fp=0x61657669) at iofclose.c:49
#1  0x0804d347 in process_outputs (channel=channel@entry=0xb6dbb470)
    at rtl_airband.cpp:431
#2  0x0804d6cd in output_thread (params=0x0) at rtl_airband.cpp:475
#3  0xb7f1af70 in start_thread (arg=0xb4231b40) at pthread_create.c:312
#4  0xb7ae9bee in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:129

Thread 6 (Thread 0xb4a32b40 (LWP 2953)):
#0  0xb7fdccb0 in ?? ()
#1  0x56a94997 in ?? ()
#2  0x84bb69bc in ?? ()
#3  0xada39eb3 in ?? ()
#4  0xbb76be8e in ?? ()
#5  0x9f4faf5c in ?? ()
#6  0x6f458740 in ?? ()
#7  0x4b665753 in ?? ()
#8  0x4795417d in ?? ()
#9  0x6db757ab in ?? ()
#10 0x9ab583bf in ?? ()
#11 0xb78db0a3 in ?? () from /lib/i386-linux-gnu/libdbus-1.so.3
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
---Type <return> to continue, or q <return> to quit---

Thread 5 (Thread 0xb5efeb40 (LWP 2952)):
Python Exception <class 'gdb.MemoryError'> NÃ£o Ã© possÃ­vel acessar a memÃ³ria no endereÃ§o 0x8: 
#0  0xb7fdccb0 in ?? ()
NÃ£o Ã© possÃ­vel acessar a memÃ³ria no endereÃ§o 0x8
#################################################

Thread 1 (Thread 0xb788f700 (LWP 3177)):
#0  __strlen_ia32 () at ../sysdeps/i386/i686/multiarch/../../i586/strlen.S:51
#1  0xb7a789bd in __GI___strdup (
    s=0x20737475 <error: NÃ£o Ã© possÃ­vel acessar a memÃ³ria no endereÃ§o 0x20737475>) at strdup.c:41
#2  0xb7f36e4f in _shout_util_strdup ()
   from /usr/lib/i386-linux-gnu/libshout.so.3
#3  0xb7f35bda in shout_set_name () from /usr/lib/i386-linux-gnu/libshout.so.3
#4  0x0804cbe8 in shout_setup (icecast=0x8058228) at rtl_airband.cp

szpajder, 2016-01-21:
Can you post your rtl_airband.conf file (without passwords, server addresses, etc)?

I did some testing, but so far I can't reproduce this crash.

jaimehempke, 2016-01-21:
OK

devices = (
  {
    index = 1;
    mode = "multichannel";
    gain = 28;
    centerfreq = 122180000;
    correction = 72;
    channels = ( 
      {
        freq = 121100000;
        outputs = ( 
          {
            type = "icecast";
            server = "#####.com.br";
            port = 8000;
            mountpoint = "121.100";
            username = "source";
            password = "######";
            name = "121.1";
            genre = "ATC";
            },
            {
            type = "file";
            directory = "/home/jaime/Fonias/Trafego";
            filename_template = "Trafego";
            continuous = false;

          }
	);
      },
      {
        freq = 121400000;
        outputs = ( 
          {
            type = "icecast";
            server = "#####.com.br";
            port = 8000;
            mountpoint = "121.4";
            username = "source";
            password = "#####";
            name = "121.4";
            genre = "ATC";
 },
            {
            type = "file";
            directory = "/home/jaime/Fonias/App12140";
            filename_template = "121.4";
            continuous = false;

          }
	);
      },
      {
        freq = 121900000;
        outputs = ( 
          {
            type = "icecast";
            server = "#####.com.br";
            port = 8000;
            mountpoint = "121.9";
            username = "source";
            password = "#####";
            },
            {
            type = "file";
            directory = "/home/jaime/Fonias/Solo";
            filename_template = "Solo";
            continuous = false;
          }
      );
      },
      {
        freq = 122400000;
        outputs = ( 
          {
            type = "icecast";
            server = "###.com.br";
            port = 8000;
            mountpoint = "live3";
            username = "source";
            password = "######";
            name = "Academia";
            genre = "ATC";
            },
            {
            type = "file";
            directory = "/home/jaime/Fonias/Academia";
            filename_template = "Academia";
            continuous = false;
            }
            );
            }, 
            {
        freq = 123400000;
        outputs = ( 
          {
            type = "icecast";
            server = "######.com.br";
            port = 8000;
            mountpoint = "123.4";
            username = "source";
            password = "#####";
       }
	);
      },
      {
        freq = 123450000;
        outputs = ( 
          {
            type = "icecast";
            server = "#####.com.br";
            port = 8000;
            mountpoint = "123.45";
            username = "source";
            password = "####";
       }
	);
      },
      {
        freq = 121500000;
        outputs = ( 
          {
            type = "icecast";
            server = "#####.com.br";
            port = 8000;
            mountpoint = "Emergencia";
            username = "source";
            password = "#####";
            name = "Emergencia";
            genre = "ATC";
            },
            {
            type = "file";
            directory = "/home/jaime/Fonias/Emergencia";
            filename_template = "121.5";
            continuous = false;

             }
          );
      } );
  },

{
    index = 0;
    mode = "multichannel";
    gain = 28;
    centerfreq = 135575000;
    correction = 72;
    channels = ( 
          {
           freq = 136825000;
        outputs = ( 
          {
            type = "icecast";
            server = "#####.com.br";
            port = 8000;
            mountpoint = "Azul";
            username = "source";
            password = "tec5643";
            name = "Azul";
            genre = "ATC";
 }
        );
      },
      {
        freq = 135750000;
        outputs = (
          {
            type = "icecast";
            server = "######.com.br";
            port = 8000;
            mountpoint = "135.750";
            username = "source";
            password = "#####";
            name = "135.75";
            genre = "ATC";
            },
            {
            type = "file";
            directory = "/home/jaime/Fonias/App13575";
            filename_template = "135.75";
            continuous = false;

            
         }
	);
      }
    );
  } );

szpajder, 2016-01-21:
The config looks good and I have no problems with it - no crashes. Tested
on two PCs - 32-bit and 64-bit Gentoo Linux (I don't have Ubuntu 14 at hand
to test on it).

If you wish, you may set up remote SSH access for me to your system and I
will see if I can help. Of course I would need root access to run
rtl_airband, so if that's OK for you, please contact me via e-mail (you can
find my address in README.md file).

szpajder, 2016-01-24:
Fixed in v2.0.1

closed 2016-01-24 by szpajder