rtl-airband/RTLSDR-Airband #16: Illegal instruction

pdostal, 2016-02-10:
Hello everyone,

I'd like to stream Trabzon International Airport from Turkey.
I've Raspberry Pi 2 model b with Raspbian Jessie.

But I'm still not able to run this program or to get some more output…

```
root@rpi:/opt/rtl_airband# sudo /usr/local/bin/rtl_airband -h
root@rpi:/opt/rtl_airband# sudo /usr/local/bin/rtl_airband -f
root@rpi:/opt/rtl_airband# /usr/local/bin/rtl_airband -f
Illegal instruction
root@rpi:/opt/rtl_airband# /usr/local/bin/rtl_airband -h
Illegal instruction
root@rpi:/opt/rtl_airband# lsusb | grep RTL
Bus 001 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
root@rpi:/opt/rtl_airband# uname -a
Linux rpi 4.1.17+ #838 Tue Feb 9 12:57:10 GMT 2016 armv6l GNU/Linux
root@rpi:/opt/rtl_airband# tail /var/log/messages
2016-02-10T14:09:41-05:00 rpi systemd[1]: Starting LSB: rtl_airband initscript...
2016-02-10T14:09:41-05:00 rpi rtl_airband[1341]: Illegal instruction
2016-02-10T14:09:41-05:00 rpi systemd[1]: Started LSB: rtl_airband initscript.
root@rpi:/opt/rtl_airband# 
```

Do you have any idea what might be wrong?

Thanks you

szpajder, 2016-02-10:
You have Raspberry Pi 1, not 2.

Note the output of "uname" command - it says the CPU is "armv6l", so you
need to compile the program with PLATFORM=rpiv1.

PLATFORM=rpiv2 is for Raspberry Pi 2, which has "armv7l" CPU.

pdostal, 2016-02-11:
You're right! It works like a charm now.

Thanks you and sorry for bad question

closed 2016-02-11 by pdostal