rtl-airband/RTLSDR-Airband #17: Support for Pi 3?

fjwarren, 2016-04-11:
I notice a switch for Pi1 vs Pi 2, but not Pi 3.  I was able to compile and run (sort of) on a Pi 3, but we are having a few issues.  I'm not sure if I have other issues or if Pi3 is just not supported yet.

I'm hoping to get a multichannel unit working to monitor all the common freqs around our airport so we desk-bound folks can a least listen to the fun on the sunny days :0  I have limited experience so far with RaPi, so any and all help / advice is appreciated.

szpajder, 2016-04-11:
RPi 3 has a bit newer CPU than RPi v2, however rtl_airband compiled with
PLATFORM=rpiv2 should work on it.

"A few issues" could mean just anything. Please describe them in detail and
post your config file (without passwords, hostnames, etc).

fjwarren, 2016-04-12:
Thanks for the quick answer.  

I believe I found my problem with my config file; some streaming password issues.  I have shortened it down to a single dongle & two channels recorded locally, and I'm seeing the behavior I expect.  I haven't seen any traffic yet (the unit is home on my bench & I'm @ work...antenna is inside in a poor location, so I'm not surprised), but I'll check it with my handheld when I get home tonight if I haven't seen traffic by then.

I did move it to a Pi2 just in case, but will try this config file in the Pi3 tonight after I verify it works & report back.

Thanks for all your work developing this...I think it's going to be a big hit at our field....

closed 2016-05-14 by szpajder