rtl-airband/RTLSDR-Airband #22: Can't make pi2 work with more than 3 dongles

mako19, 2016-06-12:
Hi, I have pi 2 with Jessie and I want to use as many dongles as it can handle (5-6). Currently I have it running 3, with CPU around 25%. I have installed RTL-Airband using manual on Code page, using "Alternatively you can pull the latest code from git" variant. My dongles are all working via good 7-port, 3,5Amp powered hub and are all functional (tested with Win7 desktop). I believe I have correct config file. When I launch the program with -f parameter I get three good waterfalls for the first 3 devices but the 4th waterfall is not there, just the list of frequences for it. All streams are written into mp3 files and the files for 4th dongle do not appear. If I delete one of devices from config file, the 4th dongle starts working; therefore, there is no mistake in the config file. Please help.
[image]
[image]
[rtl_airband.conf.txt](https://github.com/szpajder/RTLSDR-Airband/files/310844/rtl_airband.conf.txt)

mako19, 2016-06-12:
reupload of waterfall
[image]

szpajder, 2016-06-12:
Have you seen error messages like "Failed to submit transfer 12!" on the console during rtl_airband startup?

Earlier there was a report of a similar problem with three dongles. Itas has been addressed with this one-line commit: https://github.com/szpajder/RTLSDR-Airband/commit/b2c700903aa4776cb4c6b8af7c15710054973230

If it still fails for four dongles, reducing the buffer count further down, eg. to 10 or 5, might resolve the issue. Could you please edit this line in rtl_airband.cpp, recompile and check if there is any improvement?

mako19, 2016-06-12:
Thank you for response!
No, I haven't seen such messages. It starts well and works well with 3 dongles. I used "alternatively..." procedure so I didn't edit anything myself. Is version there the newest, 2.0.2? I will try your trick, thank you.

hamav8tor, 2016-06-14:
> has been addressed with this one-line commit: b2c7009

I can confirm that reducing the buffer count to 10 cures this issue for 
4 dongles.

mako19, 2016-06-14:
Thank you it seems it worked for me too

szpajder, 2016-06-14:
In current unstable branch this parameter is now configurable via global parameter "rtlsdr_buffers" in rtl_airband.conf. The new default is 10.

commit 2e0f08cb (2016-06-14): Make USB buffer count configurable

The value of USB transfer buffers can be configured via a global
parameter "rtlsdr_buffers". The new default is 10.

Closes #22
closed 2016-06-14 by szpajder