rtl-airband/RTLSDR-Airband #23: Steady SNR degradation

hamav8tor, 2016-06-14:
This may be a broader issue but I see steady SNR decrease issues when streaming ATIS/AWOS transmitters. The only reason I see it with these transmitters is that they are always transmitting and offer a stable signal for measurement purposes. The reception starts out fine then the noise will seem to steadily rise over a period of about 7-8 minutes - and the signal eventually falls below the squelch level. It gets to a point where reception becomes impossible without restarting rtl_airband. That always puts things back to normal. This means that rtl_airband needs to be restarted at least every 10 minutes to maintain reception.

Here is a video showing the phenomenon:
https://www.dropbox.com/s/aw8mupvxg0uhrvy/rtl_snr_issue.mov?dl=0

szpajder, 2016-06-14:
This is due to the fact that rtl_airband measures noise floor on the same frequency as the configured channel. Of course it cannot work properly in such cases. At least some brief periods of silence are required for it to settle down on the correct value.

I'll craft a fix for that.

hamav8tor, 2016-06-14:
Ah....makes sense. I should have realized that!

hamav8tor, 2016-07-07:
I have several setups where this can be tested once there is a patch

szpajder, 2016-07-25:
Sorry for delay, I was on vacation during recent weeks.

I tried several approaches to make auto squelch more smart, but neither worked too well and all of them would require additional computations -> more CPU power. Hence, commit https://github.com/szpajder/RTLSDR-Airband/commit/b8c2686a49bd0de317a0e430e56eaf7f8c5f04d3 in unstable branch adds a possibility to set the squelch level manually. The syntax is quite straightforward:

```
channels (

    {

        freq = 123450000;

        squelch = 200;     # some integer value greater than 0

        (...)

    }

);
```

Whenever the signal level (the first number in the textual waterfall) exceeds the given value, the squelch will open. The noise floor (second number) will show 100 constantly - it can be ignored, as the noise floor calculation is entirely skipped for channels which have squelch set manually.

All channels which don't have squelch set explicitly, will work in auto-squelch mode as before. Auto squelch opens when the signal level exceeds 3*noise level.

Moreover, earlier commit https://github.com/szpajder/RTLSDR-Airband/commit/3df2459d4dd9b010e16ae653d5e9d844e4f03c01 modifies the signal level calculation algorithm. The signal power is no longer capped at 4.5 * noise_floor, hence you will see much larger values than before (on the order of 1000 or more). As a side effect, auto squelch is now more sensitive to low power transmissions.

Please test and share your feedback.

hamav8tor, 2016-07-29:
Outstanding! This feature is very useful...and successful thus far in testing.

szpajder, 2016-08-11:
Manual squelch setting is now available in stable release 2.1.0.

Closing.

closed 2016-08-11 by szpajder