rtl-airband/RTLSDR-Airband #29: Issue building on x86_64

hamav8tor, 2016-07-22:
System: CentOS 6 x86_64

Trying to compile latest unstable branch and get this error:

PLATFORM=x86 make
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -march=native   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp: In function âvoid process_outputs(channel_t*)â:
rtl_airband.cpp:655: warning: comparison between signed and unsigned integer expressions
rtl_airband.cpp: In function âvoid demodulate()â:
**rtl_airband.cpp:886: error: âfftwf_alloc_complexâ was not declared in this scope**
rtl_airband.cpp: In function âint main(int, char**)â:
rtl_airband.cpp:1219: warning: deprecated conversion from string constant to âchar*â
rtl_airband.cpp:1221: warning: deprecated conversion from string constant to âchar*â
make: *** [rtl_airband.o] Error 1

Any ideas?

szpajder, 2016-07-22:
fftw 3.2, I suppose?

Please upgrade to 3.3.x.

hamav8tor, 2016-07-22:
That fixed it :-)

closed 2016-07-22 by hamav8tor

dranch, 2018-05-30:
I'm trying to build v3.0.1 on Centos 6.8 but I'm having a similar issue:
--
$ make PLATFORM=x86 NFM=1 PULSE=1
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/etc\" -DDEBUG=0 -march=native -DNFM -DPULSE -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp: In function ‘void demodulate()’:
rtl_airband.cpp:270: error: ‘fftwf_alloc_complex’ was not declared in this scope
rtl_airband.cpp: In function ‘int main(int, char**)’:
rtl_airband.cpp:624: error: #pragma GCC diagnostic not allowed inside functions
rtl_airband.cpp:625: warning: deprecated conversion from string constant to ‘char*’
rtl_airband.cpp:626: warning: deprecated conversion from string constant to ‘char*’
rtl_airband.cpp:627: error: #pragma GCC diagnostic not allowed inside functions
make: *** [rtl_airband.o] Error 1
--

Is fftw 3.3.x really mandatory?  I ask because the EPEL repo for Centos6 has FFTW v3.2.1

--David

szpajder, 2018-05-31:
You may replace the two `fftwf_allow_complex` calls in rtl_airband.cpp:

    fftin = fftwf_alloc_complex(fft_size);
    fftout = fftwf_alloc_complex(fft_size);

with:

    fftin = (fftwf_complex *)fftwf_alloc_complex(fft_size * sizeof(fftwf_complex));
    fftout = (fftwf_complex *)fftwf_alloc_complex(fft_size * sizeof(fftwf_complex));

You also have to comment out the two:

    #pragma GCC diagnostic ignored "-Wwrite-strings"

lines in `main()`.

dranch, 2018-05-31:
Hello Tomasz,

Thanks for replying.  I gave your changes a try (below) but it didn't resolve all the issues:
--
/usr/src/redhat/BUILD/RTLSDR-Airband-3.0.1$ diff -u rtl_airband.cpp.orig rtl_airband.cpp
--- rtl_airband.cpp.orig        2018-05-31 07:44:47.863420632 -0700
+++ rtl_airband.cpp     2018-05-31 07:47:54.905799720 -0700
@@ -267,8 +267,12 @@
        fftwf_plan fft;
        fftwf_complex* fftin;
        fftwf_complex* fftout;
+    fftin = (fftwf_complex *)fftwf_alloc_complex(fft_size * sizeof(fftwf_complex));
+    fftout = (fftwf_complex *)fftwf_alloc_complex(fft_size * sizeof(fftwf_complex));
+    /*Change out for FFTW < v3.2
        fftin = fftwf_alloc_complex(fft_size);
        fftout = fftwf_alloc_complex(fft_size);
+    */
        fft = fftwf_plan_dft_1d(fft_size, fftin, fftout, FFTW_FORWARD, FFTW_MEASURE);
 #else
        int mb = mbox_open();
@@ -621,10 +625,10 @@
 }

 int main(int argc, char* argv[]) {
-#pragma GCC diagnostic ignored "-Wwrite-strings"
+//#pragma GCC diagnostic ignored "-Wwrite-strings"
        char *cfgfile = CFGFILE;
        char *pidfile = PIDFILE;
-#pragma GCC diagnostic warning "-Wwrite-strings"
+//#pragma GCC diagnostic warning "-Wwrite-strings"
        int opt;
        char optstring[16] = "efFhvc:";
--

Now I see:
--
$ make PLATFORM=x86 NFM=1 PULSE=1
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/etc\" -DDEBUG=0 -march=native -DNFM -DPULSE -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp: In function ‘void demodulate()’:
rtl_airband.cpp:270: error: ‘fftwf_alloc_complex’ was not declared in this scope
rtl_airband.cpp: In function ‘int main(int, char**)’:
rtl_airband.cpp:629: warning: deprecated conversion from string constant to ‘char*’
rtl_airband.cpp:630: warning: deprecated conversion from string constant to ‘char*’
make: *** [rtl_airband.o] Error 1
--

szpajder, 2018-05-31:
Sorry, my mistake.

    fftin = (fftwf_complex *)fftwf_malloc(fft_size * sizeof(fftwf_complex));
    fftout = (fftwf_complex *)fftwf_malloc(fft_size * sizeof(fftwf_complex));

dranch, 2018-05-31:
Getting closer... two warnings but still one error:
--
make PLATFORM=x86 NFM=1 PULSE=1
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/etc\" -DDEBUG=0 -march=native -DNFM -DPULSE -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp: In function ‘int main(int, char**)’:
rtl_airband.cpp:629: warning: deprecated conversion from string constant to ‘char*’
rtl_airband.cpp:630: warning: deprecated conversion from string constant to ‘char*’
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/etc\" -DDEBUG=0 -march=native -DNFM -DPULSE -DWITH_RTLSDR   -c -o input-common.o input-common.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/etc\" -DDEBUG=0 -march=native -DNFM -DPULSE -DWITH_RTLSDR   -c -o input-helpers.o input-helpers.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/etc\" -DDEBUG=0 -march=native -DNFM -DPULSE -DWITH_RTLSDR   -c -o output.o output.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/etc\" -DDEBUG=0 -march=native -DNFM -DPULSE -DWITH_RTLSDR   -c -o config.o config.cpp
config.cpp: In function ‘int parse_outputs(libconfig::Setting&, channel_t*, int, int, bool)’:
config.cpp:121: error: operands to ?: have different types ‘libconfig::Setting’ and ‘const char [12]’
make: *** [config.o] Error 1
--

szpajder, 2018-05-31:
pdata->name = strdup(outs[o].exists("name") ? outs[o]["name"] : "rtl_airband");

This is not 100% correct, but newer versions of gcc can cope with it. It should be:

    pdata->name = outs[o].exists("name") ? strdup(outs[o]["name"]) : "rtl_airband";

I'll fix that.

dranch, 2018-05-31:
That worked and all warnings have been resolved!  Awesome!  Now on to see if I can get eight audio streams to send to ALSA loopback/sinks/whatever to STDOUT instead of using PulseAudio as Raspbian doesn't use PulseAudio due to various issues.  Thanks for the help!

szpajder, 2018-05-31:
Lucky me that I didn't know about these issues - so I got it running in a couple of minutes.

Anyway, you don't have to run pulseaudio server on the Pi where rtl_airband is running. You can stream to a remote Pulseaudio server running on a PC.

dranch, 2018-05-31:
Sent you an email to take this out of the Github issues area unless you'd like me to create a new issue on Github or maybe you have an email list for rtlsdr-airband support questions (groups.io, yahoo groups, whastever).