rtl-airband/RTLSDR-Airband #30: Audio Gain

dreinhold, 2016-08-18:
Looking to bump up the audio gain. Normal I do it when I convert my sdr audio files from wav to mp3 using lame. As this is already a mp3 did not want to re encode.
Any ideas in the code where I might be able to get some more audio volume.

szpajder, 2016-08-18:
For AM there is auto gain which tries to keep the audio level at 0.8 *
maximum to avoid clipping. It's in rtl_airband.cpp, around line 1186, so
you may change the upper limit from 0.8 to eg. 0.9, recompile and check if
it works better for you. I might make this constant configurable.

For NFM there is no auto gain right now.

dreinhold, 2016-08-18:
Yes forgot to say it's NFM that was looking to get the gain on.

szpajder, 2016-08-18:
The phase difference-to-amplitude translation is done in fm_quadri_demod or
polar_disc_fast  function (depending on whether you choose to use -Q option
or not). It's around line 863 and below. As the maximum possible phase
change equals pi (180 degrees) and the maximum possible amplitude is 1.0,
the scaling process is simply done by multiplying the demodulation result
by 1/pi (the M_1_PI macro). If you put a higher value in place of M_1_PI
(for example 0.4 or 0.5), you will get higher output amplitude. Don't be
tempted to increase it too much though, as it will easily cause audio
clipping. That's why I prefer to use analog method to adjust the gain -
namely, the volume knob on the amp or speakers :)

I once thought about implementing auto gain for NFM, however NFM is already
quite CPU intensive on the RPi, so I abandoned that idea in favor of
working on a few other, more often requested features.

-- 
Tomasz Lemiech

dreinhold, 2016-08-18:
Awesome, thanks for the info.
Understood I only have so much to play with until it starts clipping, I capture audio from a trunked radio system also, and want to add these couple VHF channels into the mix, but I want/need the audio to be about the same.

szpajder, 2016-11-13:
There is a new feature, called mixers, which has recently arrived into unstable. You may use it to configure a volume boost factor manually. See here for details: https://github.com/szpajder/RTLSDR-Airband/issues/40

dreinhold, 2016-11-15:
Awesome thanks for that.

closed 2016-11-15 by dreinhold