rtl-airband/RTLSDR-Airband #34: Error: gain value

spotter-ssol, 2016-08-22:
Got this error when trying to put a decimal gain value. I'm using unstable branch.
--
rtl_test reports: 
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
--
Setting gain to 36.4 as reported leads to service unable to start.
rtl_airband[2959]: Configuration error: invalid parameter type: devices.[0].gain
--
Setting gain without decimal part works ok.

szpajder, 2016-08-22:
Is this a huge problem? :-)

If you want 36.4, just type in 36 - the rtlsdr_set_tuner_gain() routine
will automatically choose the closest supported gain value.

spotter-ssol, 2016-08-22:
Sorry, I didn't knew that. :)

closed 2016-10-08 by szpajder