rtl-airband/RTLSDR-Airband #35: Audio bleeding through streams

leifskaue, 2016-08-29:
I have an RPi 1 running RTLSDR-Airband with two streams on each their own frequency 118.1 and 119.7. 
The dongle is the one from rtl-sdr.com with the R820T2 tuner. Audio is bleeding from one stream to the other - not so much that it's the same, but you can understand what they say on the other stream. It's only audible when there are transmissions on both frequencies.

Any idea what's causing this?

szpajder, 2016-08-29:
Too high gain setting, I guess.

See here on how to adjust it properly:
https://github.com/szpajder/RTLSDR-Airband/issues/8#issuecomment-207935532

leifskaue, 2016-08-29:
Seems to be the issue. I thought about it, but I didn't think 36 was that high.

closed 2016-10-08 by szpajder