rtl-airband/RTLSDR-Airband #37: Config File basic syntax

gh0straider, 2016-09-14:
I am setting up a scanner using a Raspberry Pi v2 and have Airband installed but am getting a parse error on the config file when running it. I would like a basic config file for 1 dongle running in multichannel mode. I will not be uploading to icecast instead I will be sending the feed to Broadcastify. I am trying to monitor 4 frequencies, listed below.
155.74500
155.65500
158.77500
154.38500

Thank you for any assistance.

gh0straider, 2016-09-15:
I was able to work through configuration sample file and get a working file

closed 2016-09-15 by gh0straider

trcyprkr, 2019-02-23:
Same thing happened to me.  The syntax for the frequency is nine digits with no punctuation such as
134425000

trcyprkr, 2019-02-23:
the problem i am having is "unknown output type" .  Here's my config:
devices:
({
  type = "rtlsdr";
  index = 0;
  gain = 25;
  correction = 80;
  mode = "scan";
  channels: 
	(
		{
		freqs = ( 118150000, 124750000, 134425000, 134475000, 128175000 );
		labels = ( "Tower", "Ground", "Approach", "other", "other" );
		modulation = "am";
      outputs: 
	(
		{
		type = "pulse";
		server = "127.0.0.1";
		sink = "default";
		stream_name = "De Planes";
		continuous = false;
      }
     );
    }
  );
 }
);