rtl-airband/RTLSDR-Airband #40: [Feature Request / Suggestion] - Multi channel mixing

cbgsy, 2016-10-19:
Currently using this to monitor several Marine VHF channels in multichannel mode, streaming to Icecast.

It would be useful to have the option to combine audio from 2 or more channels together to a single stream before MP3 encoding / streaming in order to reduce CPU overhead and number of streams.  This could be used for example with duplex channels with different Tx / Rx frequencies to create a single stream and avoid issues with different buffering delays between the two.

Bear in mind that the separation may well be across two different dongles on the input side (the marine VHF duplex separation is 4.6MHz, so requires 2 dongles to cover the entire band).

Mixing could either be done on a simple mono basis, or to push certain channels to either the left or right side of a stereo feed.

This could also be useful in the airband world to create a single stereo feed for an airfield with, for example, Approach on the left channel and Tower on the right to give both channels in a single feed, but still maintaining some context to the listener.

I'm currently looking to achieve this externally using something like Liquidsoap ahead of Icecast to manipulate and combine the streams, but it would be great to have flexibility at the receiver end to reduce the overhead on unnecessarily encoding and sending multiple streams.

Appreciate this would probably require some amount of re-architecture to separate the output layer from the channel definitions to give the ability to effectively output on an N:N basis rather than 1:N as currently, but hopefully should be possible?

szpajder, 2016-10-19:
I'll see what could be done about it. Stay tuned.

hamav8tor, 2016-10-19:
This would be a desirable feature. I currently use Liquidsoap to do this and when done server-side the results are not always synchronized due to the two streams arriving at slightly different times. When done from a source *nix system and an appropriate sound device (that supports simultaneous access to the device), it works great.

szpajder, 2016-11-08:
Here is some fresh stuff which seems to work (at least for me):

https://github.com/szpajder/RTLSDR-Airband/tree/unstable

There is a new concept of mixers. A mixer is a thing which has several inputs and produces a single channel with mixed audio taken from all these inputs. This channel may in turn have multiple outputs - just like an ordinary channel from the RTL dongle. It can be sent to Shoutcast or two Shoutcasts or saved to a file, or.... you name it. As there are no docs yet, here is a quick example:

```
mixers: {
  my_mixer1: {
    disable = false;
    outputs: (
        {
          type = "icecast";
          server = "my.server.com";
          port = 8080;
          mountpoint = "my_mixer1.mp3";
          name = "ATIS+TWR"
          genre = "ATC";
          username = "source";
          password = "mypassword";
        },
        {
         disable = false;
          type = "file";
          directory = "/home/user/mp3";
          filename_template = "my_mixer1";
          continuous = false;
        }
    );
  }
};
devices: (
{
  index = 0;
  gain = 25;
  centerfreq = 120000000;
  correction = 20;
  mode = "multichannel";
  channels:
  (
# ATIS
    {
      freq = 120500000;
      squelch = 80;
      outputs: (
        {
          type = "mixer";
          name = "my_mixer1";
          ampfactor = 0.3;
        }
      );
    },
# TWR
    {
      freq = 119500000;
      outputs: (
        {
          type = "mixer";
          name = "my_mixer1";
        }
      );
    }
  );
});
```
Mixers are defined in the `mixers` section. Each mixer has a name (which precedes the opening brace - `my_mixer` hereabove). After that comes a list of outputs - just like for an ordinary device.
Then comes a `devices` section with receivers and channel definitions, as usual. The new thing is the output type `mixer`, with `name` option indicating which mixer you want to connect a particular channel to. There is also an optional parameter named `ampfactor`, which is used to amplify or attenuate a channel by a fixed amount. At present, only mono mixing is supported. Stereo mixing is TODO.

There is no limit on the number of mixers you can define. They just have to be given unique names. Having in mind that each device channel can have multiple outputs, this gives quite a lot of flexibility. You can:

- route a channel to a Shoutcast, file and a mixer, all at the same time
- route a channel to more than mixer
- mix up to 32 input channels, from any dongle, into a single stream
- route mixed stream to a Shoutcast and/or a file
- adjust the volume of your single-channel streams (so far there was no volume knob, but now there is - just pipe your channel through a mixer and set `ampfactor` to some value)
- ...

Mixed channel may be saved to a file with `continuous` option set to true or false. When it's false, the stream gets recorded only when the squelch is open on any of the mixer input streams.

If any dongle happens to fall out of the USB port, its channels get disabled and disconnected from all mixers, so that the mixing process can continue with remaining input channels. If the last input of a mixer fails, the whole mixer gets disabled, ie. its outputs are closed and Shoutcast streams get disconnected.

Right now it is not possible to connect a mixer output to another mixer's input. However, if somebody provides a good use case, I may consider implementing this :-)

I have spent most of the time on development and not a lot on testing. So please use it, abuse it and share your feedback.

Oh, and during all this, I have managed to catch and fix a nasty and long-standing bug in sampling buffer management code, which caused some blocks of RTL input samples to be processed more than once. Not too relevant for AM, but quite significant for NFM. This caused a whole lot of clicking in NFM audio.

I'll add some more config examples and documentation in the coming days.

cbgsy, 2016-11-08:
Excellent - this sounds great - many thanks for doing this!

I'll download and have a play around with it tomorrow and let you know how I get on.

spotter-ssol, 2016-11-08:
I don't know about you guys, but for me this feature means "zero lag scan mode"! Thanks!

spotter-ssol, 2016-11-09:
Raspberry Pi 2 + Debian Jessie;
Center freq.: 124.0;
Monitored freq: 123.45, 124.0 124.8;
Config file parsed OK, but htop shows rtl_airband eating ~100% of one core;
Running with -f option shows waterfall running slow;
Audio is choppy, sample here: https://clyp.it/wpi2jgpk.

szpajder, 2016-11-09:
Please post your config file (without passwords, hostnames, etc).

spotter-ssol, 2016-11-09:
Sorry,  compiled for wrong platform. Need coffee. Everything is fine now.

szpajder, 2016-11-11:
Stereo mixing is now supported. Just pull the latest unstable.

New knob in the output config:
```
# TWR
    {
      freq = 119500000;
      outputs: (
        {
          type = "mixer";
          name = "my_mixer1";
          balance = -1.0;
        }
      );
    }
```
`balance` is, well, the balance knob on your HiFi set. -1.0 is full left, 0.0 is center, 1.0 is full right. I prefer 0.85 and -0.85 when using headphones - it's a bit easier on my ears than full-left and full-right setting.
Enjoy.

cbgsy, 2016-11-15:
Nice one.

Been playing around and mostly working OK for me I think.  I am occasionally getting some strange jumps in audio (on NFM receive - not tried AM yet).  Just trying variations of file vs icecast, squelch vs open etc to see if I can pin it down.  Guessing it may be a buffer drop issue, as it isn't replaced by silence - just a skip forward.  Could your NFM bug fix have introduced something else?

szpajder, 2016-11-15:
You might be dropping samples from the RTL, esp. during CPU spikes. How many dongles are you running and on what HW platform? I get occasional audio skips on RPi, but not on a PC, where the CPU is a lot more powerful.

I don't suspect this to be a side effect of the bugfix you mentioned.

cbgsy, 2016-11-16:
Appreciate that which is why I'm checking my setup.  I'll try it on the laptop too - see if it makes a difference.
I'm running 2 dongles on a RPi3 (with an externally powered USB hub), with 6 x NFM channels configured (4 + 2).  CPU is at 10% average across all cores - never much more than 30% on a single core - it's not running anything else.

szpajder, 2017-01-02:
Released in v2.3.0.

Closing.

closed 2017-01-02 by szpajder