rtl-airband/RTLSDR-Airband #42: Updating Config File to v2.x

USAFPride, 2016-11-25:
I can't seem to figure out what I need to change in my old config file to get it to work with v2.x. Any help would be appreciated:
`syslog` = 1

devices:
(
  {
    index = 0;
    gain =35;
    mode = "scan";
    correction = 52;
    channels :
    (
      {
        freqs = ( 1xxxx0000, 1yyyy0000, 1zzzz0000, 1aaaa0000, 1bbbb0000, 1cccc0000 );
        server = "xxx.xxx.xxx.xxx";
        port = yyyy;
        mountpoint = "aaaa";
        username = "bbbb";
        password = "cccc";
        name = "ddddd";
        genre = "ATC";
      }
    );
}
);
`

USAFPride, 2016-11-25:
The error I get is:
Configuration error: mandatory parameter missing: devices.[0].channels.[0].outputs

szpajder, 2016-11-25:
As we now support multiple outputs for each channel, output-related settings are now in a separate section named "outputs". So it goes like this:
```
syslog = 1;
devices:
(
 {
  index = 0;
  gain = 35;
  mode = "scan";
  correction = 52;
  channels :
  (
   {
    freqs = ( 1xxxx0000, 1yyyy0000, 1zzzz0000, 1aaaa0000, 1bbbb0000, 1cccc0000 );
    outputs:
    (
     {
       type = "icecast";
       server = "xxx.xxx.xxx.xxx";
       port = yyyy;
       mountpoint = "aaaa";
       username = "bbbb";
       password = "cccc";
       name = "ddddd";
       genre = "ATC";
     }
    );
   }
  );
 }
);
```
Refer to `rtl_airband.conf.example` file for more examples.

USAFPride, 2016-11-25:
Thanks.  Missed that every time I went through it.

closed 2016-11-25 by USAFPride