rtl-airband/RTLSDR-Airband #43: Output to loopback

chris021, 2016-12-03:
Hi,

I would really love to be able to select a loop back audio device as a channel output option.  This would allow other streaming options like mumble to be used outside of RTLSDR-Airband.

szpajder, 2016-12-04:
Well, my current plan is to add support for outputting to ordinary ALSA
audio device(s). This is TODO.

Later on, we might think of some more fancy audio routing stuff, like
Pulseaudio or JACK - if that's what you mean by "loop back audio device".
Do you know if Mumble can work with these?

chris021, 2016-12-07:
Outputting to ALSA would be fine for my purposes. 👍 

I Pipe prt_fm into | aplay -r 48k -f S16_LE -t raw -c 1 -D hw:Loopback,0,0

szpajder, 2017-10-21:
How about doing this with Pulseaudio, which is supported in rtl_airband 2.4.0?

Quick'n'dirty example:

Put this into `/etc/pulse/default.pa` on the machine acting as a PA server:

    load-module module-pipe-sink file=/tmp/fifo_sink rate=8000 channels=1 format=float32le

If you are going to run rtl_airband on another machine, enable this as well (adjust the acl, so that it allows rtl_airband to connect):

    load-module module-native-protocol-tcp auth-ip-acl=127.0.0.1;10.0.0.0/24

Put this in your `rtl_airband.conf`:
```
    outputs: {
        {
            type = "pulse";
            server = "pa-server.name.com";
            stream_name = "My fancy stream";
            sink = "fifo_output";
        }
    (...)
    }
```
(you can omit the `server` parameter if PA is running on localhost; rtl_airband will then use UNIX socket to connect to it).

Restart PA server, start rtl_airband, check the logs if it has opened the stream successfully.

Play the stream with:

    aplay -r 8000 -f FLOAT_LE -t raw -c 1 </tmp/fifo_sink

or pipe it to any program you wish.

barjac, 2018-01-23:
Could you please expand a little on the above?
I have installed rtlsdr_airband in the past using icecast and using pulseaudio on one machine without problems, however now I want to run rtl_airband on a RPi2 near the antenna and want to use the pulseaudio server to serve it to other machines. I have installed exactly as per the README.md and added the systemd unit file. I see repeated errors:
pulse: <default_server>:connection failed: Connection refused
pulse: <default_server>: disconnecting context for stream "133.800 MHz"
pulse: <default_server>:  failed to connect: Invalid argument
pulse: <default_server>: connecting... 
Then the same repeated over.
There is no mention of the commands in the previous comment in the README.md or anywhere else that I can find - what am I missing and what needs to be applied to which machines to get this working?

barjac, 2018-01-23:
rtl_airband.conf is currently as follows:

devices:
({
  index = 1;
  gain = 30;
  correction = 59;
  mode = "scan";
  channels:
  (
    {
      freqs = ( 133800000, 134425000, 121850000, 118575000, 133125000 );
      labels = ( "GCC TWR", "GCC TWR2", "GCC GND", "GCC APP", "LBA TWR" );
      outputs: (
        {
	  type = "pulse";
          name = "GCC Tower + GCC Tower2 + GCC Ground + GCC Approach + LBA TWR";
          genre = "ATC";
	  send_scan_freq_tags = false;
        },
        {
          type = "file";
          directory = "/home/pi/ATC_recordings";
          filename_template = "TWR+TWR2+GND+APP+LBA";
        }
      );
    }
  );
 }
);

Note the index=1 is because I have two rtl dongles on the machine - one for ADS-B/MLAT and one for rtl_airband.

szpajder, 2018-01-23:
Sorry about the docs, I'll make it better. There is an example of pulse
output in config/two_dongles_multiple_outputs.conf file.

rtl_airband acts as a Pulseaudio client, while the machine with the sound
card and speakers is the Pulseaudio server. You need to add the "server="
option to the rtl_airband's pulse output config section to set the host
name or IP address of the machine where PulseAudio server is running.
rtl_airband will connect to the server and stream the audio there and the
server will play it. If you omit the "server=" option, then rtl_airband
attempts to connect to a default server (which is most often, localhost).
You get "connection refused" messages because there is no Pulseaudio server
running on your RPi.

Pulseaudio daemon usually does not need to be started by hand. If you have
a graphical Linux desktop session running (eg. KDE, Gnome, etc), then it's
probably already running (verify with "ps xuaw|grep pulse").

By default, Pulseaudio daemon does not listen for inbound connections over
network. To enable this, edit /etc/pulse/default.pa and add or uncomment
the following line:

    load-module module-native-protocol-tcp auth-ip-acl=127.0.0.1;
192.168.10.0/24

The auth-ip-acl parameter is a semicolon-separated list of hosts and
networks which should be allowed to connect, so put your LAN subnet there.
Then restart Pulseaudio server (or just kill it - it shall get restarted
automatically on the first attempt to play any sound). Ensure it's
listening on port 4713 ("netstat -an|grep 4713" should show a socket in a
state of LISTEN).

Of course make sure that no firewall is blocking connections from the Pi to
port 4713 on the server machine. Basic stuff.

Once you start rtl_airband, it shall connect to the server and you should
get audio played straight away. No need to start any player program, etc.
And you get a per-stream slider in your mixer app.

Note: "genre" and "send_scan_freq_tags" options do not work for pulse
outputs (they are ignored), while the Pulse stream name is set with
"stream_name" option, not "name".

barjac, 2018-01-23:
Many thanks for your rapid reply, terminology was confusing me. I was thinking that the PA 'server' would be on the RPi and that I was missing PA packages on the RPi. Now I understand.
I was hoping to connect from two different machines, laptop and desktop (independently), however  reading your explanation it seems that only one server= option is possible, or am I mistaken?
I will test all this later now I have a better understanding.

szpajder, 2018-01-23:
Yes, the key thing to understand with Pulseaudio is that it is not the
laptop and desktop who connects to rtl_airband, but the reverse.
rtl_airband connects to whoever has the sound card and speakers. The
implication is that if you turn your laptop off, rtl_airband will notice
the connection loss and will repeatedly try to reconnect, trashing your log
file with errors. But the upside of that is once you start the laptop
again, streaming will resume automatically, without any manual action.
Pretty convenient.

Only one server= option is allowed per pulse output. If you want to stream
the same channel to more than one machine, just configure multiple pulse
outputs under relevant channel:

channels: (
{
    freq = 123450000;
    outputs: (
    {
        type = "pulse";
        server = "laptop.ip";
        stream_name = "something";
    },
    {
        type = "pulse";
        server = "desktop.ip";
        stream_name = "something";
    }
    );
}
);

barjac, 2018-01-24:
Many thanks - I have it working now and fr24feed is also working with the other dongle.
RPi2 seems quite happy:
14:23:49 up 15:13,  2 users,  load average: 0.78, 0.62, 0.63
I can see above how you have created a second stream to another PA server for one channel but can this be done in my case where I am scanning?

szpajder, 2018-01-24:
>
> I can see above how you have created a second stream to another PA server
> for one channel but can this be done in my case where I am scanning?
>

Yes.

barjac, 2018-01-24:
Wow that was quick!
So I need to repeat the whole channels: section in the config but with the new server IP etc., is that correct?

szpajder, 2018-01-24:
>
> So I need to repeat the whole channels: section in the config but with the
> new server IP etc., is that correct?
>
No. You can only have a single channels: ( ) list per device. What's more,
when you are scanning, you can only have a single channel { } section in
that list. But you can have any number of output { } sections inside
outputs: ( ) list.

There is an example of scanning in
https://github.com/szpajder/RTLSDR-Airband/blob/master/config/basic_scanning.conf
- it has two outputs, first one to Icecast, second one to a file. In your
case there will be two outputs of type "pulse" instead.

barjac, 2018-01-24:
Great thanks - finally sinking in - takes a while at my age ;)

szpajder, 2018-02-10:
Closing this, as the requested feature is already available since 2.4.0 via Pulseaudio pipe sink.

closed 2018-02-10 by szpajder

kenventions, 2024-12-30:
Regarding your comment "And you get a per-stream slider in your mixer app." Will pavucontrol allow me to mix the rtl_airband audio with the microphone audio plugged into a USB audio dongle that's plugged into the Raspi?

kinglevel, 2024-12-30:
> Regarding your comment "And you get a per-stream slider in your mixer app." Will pavucontrol allow me to mix the rtl_airband audio with the microphone audio plugged into a USB audio dongle that's plugged into the Raspi?

pulseaudio is only available as an output.

If you mix it outside of rtl_airband with pulse, and then stream it to lets say an icecast server, yes.

But no analog input as an device as you can see under DEVICES in the documentation here:

https://github.com/charlie-foxtrot/RTLSDR-Airband/wiki/Configuration-essentials

kenventions, 2024-12-30:
Yes, you're "no analog input" was useful feedback - my plan B is to mix the analog input outside of rtl_airband. Since rtl_airband is an audio client sending audio to pulseaudio (localhost), I'll have to figure out a way to also send my USB analog audio to pulseaudio as a 2nd audio client. PAVUCONTROL looks like a mixer for pulseaudio - I'll let you know if I make any progress.