rtl-airband/RTLSDR-Airband #44: Odroid C2 compile problem.

starvald, 2016-12-25:
**Have it running on my pi3 but i'd like to get it running on my odroid if possible.**

odroid@odroid64:~/compiles/RTLSDR-Airband$ PLATFORM=armv7-generic NFM=1 make
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -march=armv7-a -mfpu=neon-vfpv4 -mfloat-abi=hard -ffast-math  -DNFM   -c -o rtl_airband.o rtl_airband.cpp
g++: error: unrecognized command line option ‘-mfpu=neon-vfpv4’
g++: error: unrecognized command line option ‘-mfloat-abi=hard’
<builtin>: recipe for target 'rtl_airband.o' failed
make: *** [rtl_airband.o] Error 1

**googled and tried adjusting the makefile, different error this time.**

odroid@odroid64:~/compiles/RTLSDR-Airband$ PLATFORM=armv7-generic NFM=1 make
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -march=armv8-a+crc -mtune=cortex-a53 -ffast-math  -DNFM   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp:38:23: fatal error: xmmintrin.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'rtl_airband.o' failed
make: *** [rtl_airband.o] Error 1

**xmmintrin.h doesn't exist unfortunately.**

uname -a 
Linux odroid64 3.14.79-97 #1 SMP PREEMPT Fri Dec 23 10:32:12 BRST 2016 aarch64 aarch64 aarch64 GNU/Linux

cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.1 LTS"

szpajder, 2016-12-25:
Your adjusted CFLAGS look good. Additionally, please edit the `rtl_airband.cpp` file and replace all occurences of `__arm__` with `__aarch64__` and let me know if it works.

starvald, 2016-12-25:
**Thanks for replying...**

odroid@odroid64:~/compiles/RTLSDR-Airband$ PLATFORM=armv7-generic NFM=1 make
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -march=armv8-a+crc -mtune=cortex-a53 -ffast-math  -DNFM   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp:38:23: fatal error: xmmintrin.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'rtl_airband.o' failed
make: *** [rtl_airband.o] Error 1

**Still complains about missing xmmintrin.h unfortunately.**

szpajder, 2016-12-25:
Can you show the output of `cpp -dM </dev/null|grep aarch` ?

starvald, 2016-12-25:
odroid@odroid64:~/compiles/RTLSDR-Airband$ cpp -dM </dev/null|grep aarch
#define __aarch64__ 1

szpajder, 2016-12-25:
Alright, then replace `__arm__` with `aarch64` instead of `__aarch64__` in rtl_airband.cpp.

For example, around line 27 you should have `#ifdef aarch64` instead of `#ifdef __arm__`.

starvald, 2016-12-25:
arghh dammit noticed it modified the cpp file earlier on the pi3 rather than the odroid.

`changing __arm__  to  __aarch64__ works,`

many thanks.

szpajder, 2016-12-25:
Thanks for confirmation. I'll push a fix for this into the next release.

szpajder, 2016-12-30:
Could you please pull this tree:

https://github.com/szpajder/RTLSDR-Airband/tree/unstable

and check whether it compiles and runs on C2?

Use `PLATFORM=armv8-generic` to compile. Thanks in advance.

starvald, 2016-12-30:
```
odroid@odroid64:~/compiles/RTLSDR-Airband$ PLATFORM=armv8-generic NFM=1 make
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -DNFM   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp:37:23: fatal error: xmmintrin.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'rtl_airband.o' failed
make: *** [rtl_airband.o] Error 1

```

szpajder, 2016-12-30:
Hah, I screwed that one ;) It's __aarch64__, not __arch64__.

Please pull and try again.

starvald, 2016-12-30:
Great, compiles fine now.
thank you.

szpajder, 2017-01-02:
Fixed in v2.3.0.

Closing.

closed 2017-01-02 by szpajder