rtl-airband/RTLSDR-Airband #46: Use ogg and/or change bitrate

ianatkin80, 2017-01-12:
Is it possible to use .ogg as the streaming format with icecast? I'd also like to be able to vary the sample rate and bit rate.

szpajder, 2017-01-12:
These things are currently hardcoded. You can change them by editing the
code and recompiling.

Do you need these knobs just because, or you find the audio quality with
current encoding params unacceptable?

ianatkin80, 2017-01-12:
Mostly "just because".  I usually listen to streams on a 4G connection so I was interested to see if I could reduce the bitrate and still get acceptable quality. I also wanted to explore the effect of sample rate on quality. 

What is the hardcoded mp3 bitrate out of interest?

szpajder, 2017-01-12:
Bitrate is set to 16 kbps, however this is an average, because VBR is
enabled. When the channel is silent, it produces ca 10 kbps.

If you wish, you may fiddle with various `lame_set_*`  function params in
`airlame_init()` routine (it's in `output.cpp` file). For example, there is
VBR quality currently set to 7, so you may try to lower it. Do not change
`WAVE_RATE`, because several important internals depend on it, so things
may break. You may change `MP3_RATE` instead - this is the sample rate of
the output stream (it's currently 8000, I'm not sure if liblame allows any
lower values).

Eventually I might add some config knobs for adjusting these things,
however I consider this a bit of low priority, as there are more important
features pending on the TODO list right now.

ianatkin80, 2017-01-12:
Thank you for the information, much appreciated ☺

closed 2017-01-12 by ianatkin80