rtl-airband/RTLSDR-Airband #47: Issue trying to make for armv8 platform

brantel, 2017-01-25:
Noticed you had a new option for 64bit ARM8 processors.  I have a RPi 3 and tried to make the install but it fails trying to find the fftw3.h file.

`pi@liveatc:~/RTLSDR-Airband $ sudo PLATFORM=armv8-generic make
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=armv8-a+crc -mtune=cortex-a53 -ffast-math   -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp:33:19: fatal error: fftw3.h: No such file or directory
 #include <fftw3.h>
                   ^
compilation terminated.
<builtin>: recipe for target 'rtl_airband.o' failed
make: *** [rtl_airband.o] Error 1
`

Any ideas?

mollerdaniel, 2017-01-25:
> FFTW3 library is needed in this case

`sudo apt-get install libfftw3-dev`

szpajder, 2017-01-25:
You can install libfftw3-dev and compile with armv8-generic, however it will not use GPU for FFT and the performance will probably be mediocre.

Instead, pull this branch: https://github.com/szpajder/RTLSDR-Airband/tree/rpiv3 . Compile with `PLATFORM=rpiv3 make` and let me know if it works (ie. if it does not hang the Pi and the CPU usage is reasonable, not near 100%).

brantel, 2017-01-25:
@mollerdaniel 

Thanks, I should have looked deeper.  I am just a noob at this Linux/Raspbian stuff.

brantel, 2017-01-26:
> You can install libfftw3-dev and compile with armv8-generic, however it will not use GPU for FFT and the performance will probably be mediocre.

> Instead, pull this branch: https://github.com/szpajder/RTLSDR-Airband/tree/rpiv3 . Compile with PLATFORM=rpiv3 make and let me know if it works (ie. if it does not hang the Pi and the CPU usage is reasonable, not near 100%).

@szpajder 

Thanks, wow you are fast creating branches.  I tried it and it seems to work fine without the issues you wanted a report on.  Attached is my screenshot from HTOP while running this compile.  

FYI:  I am running a RPi3 with Jessie without overclocking.  I have 3ea of these [dongles](https://www.amazon.com/gp/product/B01HA642SW/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1)  with this [USB HUB](https://www.amazon.com/gp/product/B005NGQWL2/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1) and I am uploading 3 separate streams to LiveATC.

[image]

Since I am just learning, can you explain what benefit there would be to compiling as rpiv3 vs rpiv2?

szpajder, 2017-01-26:
At first I thought that you can't compile it on RPi3 at all, but I assume
that it worked when compiled with PLATFORM=rpiv2, but you just wanted to
check the armv8-generic build, right?

armv8-generic build is meant for 64-bit platforms, which do not have
Broadcom VideoCore GPU, but their CPU is powerful enough to handle the work
(see https://github.com/szpajder/RTLSDR-Airband/issues/44). On the Pi it
will perform suboptimally, because the GPU won't be utilized at all.

If your compiler does not complain about -march=armv7-a or -mfpu=neon-vfpv4
flag, then it's fine to compile with PLATFORM=rpiv2. Performance difference
as compared with PLATFORM=rpiv3 would probably be insignificant - but I
might be wrong, because I don't have RPi3 at hand, so you might verify this
by yourself if you wish.

mollerdaniel, 2017-01-26:
I'm currently running PLATFORM=rpiv3 from the rpiv3 branch on a RPi3, no issues at all so far.
CPU usage is the same, but worth mentioning is that I only have a simple scan-setup so far.

brantel, 2017-01-26:
@szpajder 

Sorry for the confusion.  Yes I was able to compile and have been using the RPi3 with PLAFORM=rpiv2 for some time now.  That worked just fine.

I was as you said just wanting to check out the new armv8=generic build.  I had no idea about what the differences were.  Thanks for the explanation.  Makes sense to me.

I have been running on PLATFORM=rpiv3 for about 12 hours now with no negative issues I can see.  Performance seems to be good.  I do see some differences in how the CPU cores are being used in HTOP.

szpajder, 2017-01-26:
Thanks everybody for the feedback. However, if the program runs OK on Pi 3 when compiled with PLATFORM=rpiv2, then I would prefer not to add PLATFORM=rpiv3 to the mainline, because it makes the build process even more confusing for the user. I'd rather move to some automated build configuration (like autotools or cmake) instead of further extending the clutter we have now...

HagarViking, 2017-01-26:
Automated sounds like a good idea. 
And I must say you are doing one heck of a great job on this @szpajder!
It is very much appreciated! Thank you!

brantel, 2017-01-26:
I agree!  Great job!  Automated sounds great!

marcus-aa, 2017-02-05:
Hi, thanks for the interesting sw.

I have installed main release on an Rpi3, but trying to configure using 

PLATFORM=rpiv2 make 

results in :

compilation terminated.
<builtin>: recipe for target 'gpu_fft_base.o' failed
make[1]: *** [gpu_fft_base.o] Error 1
make[1]: Leaving directory '/root/RTLSDR-Airband/hello_fft'
makefile:84: recipe for target 'hello_fft' failed
make: *** [hello_fft] Error 2

Also tried armV8-generic (after installing libfftw3-dev)

PLATFORM=armv8-generic  make
g++   rtl_airband.o output.o config.o util.o mixer.o  -lrt -lm -lvorbisenc -lmp3lame -lshout -lpthread -lrtlsdr -lconfig++ -lfftw3f -o rtl_airband
rtl_airband.o: In function `demodulate()':
/root/RTLSDR-Airband/rtl_airband.cpp:319: undefined reference to `mbox_open()'
/root/RTLSDR-Airband/rtl_airband.cpp:321: undefined reference to `gpu_fft_prepare(int, int, int, int, GPU_FFT**)'
/root/RTLSDR-Airband/rtl_airband.cpp:399: undefined reference to `samplefft'
/root/RTLSDR-Airband/rtl_airband.cpp:418: undefined reference to `gpu_fft_execute(GPU_FFT*)'
/root/RTLSDR-Airband/rtl_airband.cpp:367: undefined reference to `gpu_fft_release(GPU_FFT*)'
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'rtl_airband' failed
make: *** [rtl_airband] Error 1

marcus-aa, 2017-02-05:
I'm probably missing something simple....

Same problem with rpiv3 branch:

~/RTLSDR-Airband# PLATFORM=rpiv3  make
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a+crc -mtune=cortex-a53 -ffast-math   -c -o rtl_airband.o rtl_airband.cpp
make -C hello_fft
make[1]: Entering directory '/root/RTLSDR-Airband/hello_fft'
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -march=armv8-a+crc -mtune=cortex-a53 -ffast-math -I..   -c -o gpu_fft_base.o gpu_fft_base.c
gpu_fft_base.c:29:22: fatal error: bcm_host.h: No such file or directory
 #include "bcm_host.h"
                      ^
compilation terminated.
<builtin>: recipe for target 'gpu_fft_base.o' failed
make[1]: *** [gpu_fft_base.o] Error 1
make[1]: Leaving directory '/root/RTLSDR-Airband/hello_fft'
makefile:92: recipe for target 'hello_fft' failed
make: *** [hello_fft] Error 2

szpajder, 2017-02-05:
The key information is here:

    gpu_fft_base.c:29:22: fatal error: bcm_host.h: No such file or directory

Most probably the `/opt/vc/include` directory and files are missing. Are you running Raspbian? If yes, then run `sudo rpi-update` and it should get installed. Then reboot and try again with the rpiv2 or rpiv3 build.

`PLATFORM=armv8-generic` build will succeed on the Pi3, however each time you decide to switch to another `PLATFORM` value, you have to delete files produced by the previous build by running `make clean`. Then try again.

marcus-aa, 2017-02-05:
Great, "make clean" solved the issue of re-compiling with armv8-generic, thanks for the tip.

I use DietPi, a lean distribution of Debian, and there is no rpi-update. I tried apt-get update but didn't help.

Googling indicated that the bcm_host file belongs to  libraspberrypi-dev or raspberrypi-kernel-headers, so I installed those packages and then it found the file.

It does throw these warnings but maybe works as intended anyway?

gpu_fft_base.c: In function ‘unsigned int gpu_fft_base_exec_direct(GPU_FFT_BASE*, int)’:
gpu_fft_base.c:65:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (q=0; q<num_qpus; q++) { // Launch shader(s)
                ^
gpu_fft_base.c:72:50: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (((base->peri[V3D_SRQCS]>>16) & 0xff) == num_qpus) break; // All done?
                                                  ^
gpu_fft_base.c:54:17: warning: unused variable ‘t’ [-Wunused-variable]
     unsigned q, t;
                 ^

Anyway, I suggest adding the "make clean" tip to the instruction and  maybe mention 
 libraspberrypi-dev
 raspberrypi-kernel-headers
dependency for those that don't use Raspbian.

closed 2017-02-09 by brantel