rtl-airband/RTLSDR-Airband #48: Enhancement Request: Decrease time to rebuild connection in the event of backlog buffer overrun

brantel, 2017-01-26:
Not sure if this is even possible but could an enhancement be added to decrease the time it takes to rebuild a connection in the event of a backlog buffer overrun.  Specifically to an icecast server.  More specifically to the LiveATC icecast server.

Currently it seems that the LiveATC icecast server does not drop the session very quickly or it never actually gets the disconnect request.  This results in RTLSDR-Airband trying to reconnect while the original session is still active on the LiveATC icecast server.  Most times it takes about 3 cycles for the connection to get going again.  This results in about 30 seconds being lost for every backlog buffer overrrun.

hamav8tor, 2017-01-26:
This may have to do with the source timeout in Icecast. I dropped it just a short while ago to a value (10 seconds) that may help deal with this issue. -Dave

szpajder, 2017-01-26:
Thanks Dave, this is certainly much better solution than anything I could do on the client side. My feed had one exceeded backlog event since then and it reconnected after four seconds.

@brantel - backlog occurs when there packets get lost. In order for the FIN packet to be processed by the server, it has to arrive in sequence, that is, all the earlier packets which got dropped, have to be successfully retransmitted and delivered before. Until the server disconnects the source (either by timing out or receiving FIN from the source), the client won't be able to connect again, no matter how hard it tries.

brantel, 2017-01-26:
Thanks @hamav8tor and @szpajder ,

I estimated that it would take some collaboration with @hamav8tor on this one to reduce the time lost.  Thank you both for the support!

It appears in my case, the server is not getting the disconnect message "FIN packet" for some reason and to get the connection back up and running required RTLSDR-Airband to keep trying while waiting on the timeout to expire on the server side so it could log back in.

The mystery as to why I see so many backlog buffer overruns and the apparent lost FIN packets are something I am still working on with my ISP.  I have worked around this issue 99% of the time with a constant running "ping -f" which keeps things happy most of the time.  I don't want to leave it that way due to the fact that it waste bandwidth and processor cycles.  My RPi3 is directly connected to my ISP's ONT without any sort of consumer router in between.  I have tried to uplink on two different service locations with the same ISP and both have the same issue. I do not have the issue when I try it at my house with a totally different ISP.  This leads me to believe the problem is somewhere within the ISP.  I have a support ticket in with their engineers.  I can 100% reproduce the issue at will so I won't give up on them until they find the problem.

brantel, 2017-01-26:
@hamav8tor 

I have noticed a significant decrease in the time/retries it takes for RTLSDR-Airband to reconnect since you made your change.  Thanks!

brantel, 2017-01-27:
@hamav8tor 

Dave,

After letting it run overnight, I only had 2 backlog buffer overruns on the most active channel of my feed. (ping -f running in the background) and I can confirm the changes you made yesterday have cut the time it takes for RTLSDR-Airband to reconnect has been reduced from about 30 seconds or more to 3-4 seconds!

Thank you and @szpajder  very much!

brantel, 2017-02-09:
My ISP issues seem to be resolved.  No longer have to run the ping -f to the LiveATC server in parallel with the feeds to keep from getting many backlog buffer overruns per hour.

closed 2017-02-09 by brantel