rtl-airband/RTLSDR-Airband #49: FM Support?

trs79, 2017-02-26:
Does the software support FM frequencies or is it strictly for AM?

szpajder, 2017-02-26:
The software supports any frequency which your receiver can be tuned to. For RTLSDR dongles, this translates to the range between 24 and 1700 MHz (approximately). Both AM and NFM modulation is supported regardless of frequency. Broadcast FM is not supported at all, because it's a different modulation scheme (wideband FM, not narrowband FM).

trs79, 2017-02-27:
Thanks for the information. Are there any plans to support WFM in the future? Are there any other projects that utilize the GPU that receive WFM? In the past I've used a si470x FM tuner USB dongle, but I've been wanting to try out a SDR that didn't use too much CPU.

szpajder, 2017-02-27:
The channelization method implemented in rtl_airband is suitable for
simultaneous reception of multiple narrowband channels. It doesn't work for
wideband channels (where "wide" means "several times wider than the
bandwidth of a single FFT bin"). That's why rtl_airband will not support
WFM in the foreseeable future.

There are other tools which can do WFM - for example rtl_fm (bundled with
rtl-sdr library) can demodulate WFM and it performs all calculations using
fixed point arithmetic, so the performance is decent. Audio data is written
to standard output, so it's possible to stream it via Icecast by piping it
to some streaming software which can read audio from standard input.

closed 2017-05-16 by szpajder