rtl-airband/RTLSDR-Airband #50: Arbitrary band monitoring and logging?

hellyberry, 2017-03-14:
Is RTLSDR-Airband currently capable to use a single DVBT-Stick to monitor and process e.g. ADSB and ACARS bands and messages concurrently or any other band and protocol there's a command-line tool to process from a pipe?

There's the AM-radio config example, but does this work with different protocols?
If not, can it made to?

szpajder, 2017-03-14:
RTLSDR-Airband is meant solely for voice AM and NFM channels. There are other good tools for digital aviation protocols. There is dump1090 for ADS-B, acarsdec for ACARS and dumpvdl2 for VDL-2.

What's the point of processing ADS-B and ACARS on a single stick? These are totally different frequency bands, so two different antennas are required. What's more, the stick would need to be constantly retuned between bands, which means you would lose a lot of messages, especially ADS-B which are sent very often.

I'd rather recommend running ACARS and ADS-B on separate sticks. They are very cheap, after all.

hellyberry, 2017-03-19:
I see, so two sticks are required. Thank you for your analysis.

closed 2017-05-16 by szpajder