rtl-airband/RTLSDR-Airband #51: Use Mali GPU

jmceara, 2017-03-16:
Hi.
Is possible to use Mali GPU for FFT? If so, how?
I have an specific hardware with A20 CPU (Mali400 as GPU) and I would like to run rtl_airband ins this board, but it's consuming about 80~90% using main CPU for FFT. Since this CPU has a Mali400 GPU, I would like to use for FFT calculations.

szpajder, 2017-03-16:
Right now only Broadcom Videocore is supported for FFT offloading.

jmceara, 2017-04-14:
Anybody interested in optimizing for Mali400 GPU (A20 processor)? Pay service.

Strykar, 2019-03-13:
Is OpenCL required on the GPU for FFT offloading?
I was under the impression that Mali Utgard isn't a great choice for that sort of work. No easy way to do compute directly on it, ruling out Mali 400/450.

danmilon, 2019-03-14:
I'm not very knowledgeable in regards to FFT and GPUs, but I was thinking how we could make it more generic by implementing it on top of opengl which is supported on SBCs, albeit not very recent versions of it.

For example [GLFFT](https://github.com/Themaister/GLFFT) looks promising, or this [how-to guide](https://arm-software.github.io/opengl-es-sdk-for-android/ocean_f_f_t.html). But they're both targeting OpenGL ES 3.1, and raspberry supports only up to 2.0. OTOH [Orange PI 3](http://www.orangepi.org/Orange%20Pi%203/) supports 3.1.

Maybe we can add an extra platform build target for OpenGL ES 3.1?
Someone who knows better call tell us if this idea is worth pursuing?

Strykar, 2019-03-14:
@danmilon Yea, bummer, my C2 mali 450 supports only [OpenGL ES 1.1/2.0](https://developer.arm.com/products/graphics-and-multimedia/mali-gpus/mali-450-gpu), this doesn't look promising.

charlie-foxtrot, 2023-04-21:
No plans to add additional GPU support

closed 2023-04-21 by charlie-foxtrot