rtl-airband/RTLSDR-Airband #52: Raspberry Pi Zero W - Nothing happens

DanV410, 2017-03-19:
Nothing happens and no errors or log messages generated when attempting to run on Raspberry Pi Zero W. rtl_tcp and rtl_test seem to work fine. Did see an error while compiling, attaching output for review.

Tried running with -v, -h and -f options. Nothing happens with any option.

Tried all three solutions under Troubleshooting, nothing needed changing.

[pizero_install.txt](https://github.com/szpajder/RTLSDR-Airband/files/852853/pizero_install.txt)

szpajder, 2017-03-19:
If it's Raspbian with a default rsyslogd config, I would rather look for
rtl_airband logs in /var/log/messages.

Have you actually prepared a valid configuration file? If yes - can you
post it (without passwords, hostnames, etc)?

DanV410, 2017-03-19:
Sure. Nothing get posted in /var/log/messages either. Here is my config, I counted brackets and parenthesis, so I hope I'm not making a silly mistake. I think I did the same thing on my Pi3 and that worked fine.

# This is a minimalistic configuration file for RTLSDR-Airband.
# Just a single RTL dongle with two AM channels in multichannel mode.
# Each channel is sent to a single Icecast output. Settings are
# described in reference.conf.

devices:
({
  index = 0;
  gain = 40;
  centerfreq = 460000000;
  correction = 0;
  channels:
  (
    {
      modulation = "nfm";
	  freq = 460600000;
      outputs: (
        {
          type = "file";
          directory = "/home/dan/recordings";
          filename_template = "EMS+";
        }
      );
    },
  );
 }
);

DanV410, 2017-03-19:
If needed, I can start with a new image, and log the whole ssh session as I install rtl_sdr and then rtl_airband.

szpajder, 2017-03-19:
Here is what I get with your config file:
```
szpajder@helix:~/stuff/rtl-sdr/rtl_airband/RTLSDR-Airband $ ./rtl_airband -c ../dan.conf
Error while parsing configuration file ../dan.conf line 20: syntax error
```
You have a stray comma at line 19 (ie. fourth from the bottom).

DanV410, 2017-03-19:
OK, removed that, thanks. Actually replaced rtl_airband.conf with the file from my Pi3 that works. Still the same deal, no output at all, nothing logged, no errors.

szpajder, 2017-03-19:
Is the program actually running or it exits instantly? Can you see the
process in the 'ps xuaw' output?

If it hangs, try running it with strace to see what syscall it stops on.

szpajder, 2017-03-19:
And by the way, RPi Zero has Broadcom BCM2835, so it's ARMv6, not v7. So
you should compile with PLATFORM=rpiv1, not rpiv2.

DanV410, 2017-03-19:
Ugh, that was it, sorry. I don't know why I had it in my head that the Pi Zero W used the same processor as the Pi 3. Thank you for your help, and thank you for making the software available.
Dan

closed 2017-03-19 by DanV410