rtl-airband/RTLSDR-Airband #53: No instruction to install librtlsdr-dev before compilation

ronan-fontenay, 2017-03-24:
Following the instruction to install on Raspbian on a Pi Zero : 
`
pi@raspberrypi:~/RTLSDR-Airband $ PLATFORM=rpiv1 make
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -DUSE_BCM_VC -I/opt/vc/include  -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk -mfpu=vfp -ffast-math    -c -o rtl_airband.o rtl_airband.cpp
rtl_airband.cpp:66:21: fatal error: rtl-sdr.h: No such file or directory
 #include <rtl-sdr.h>
                     ^
compilation terminated.
<builtin>: recipe for target 'rtl_airband.o' failed

Corrective action : 
> pi@raspberrypi:~/RTLSDR-Airband $ sudo apt-get install librtlsdr-dev

Should we not mention in ?

szpajder, 2017-03-25:
README.md has instructions on how to install librtlsdr from Git. This is because once upon a time this lib was not packaged in Raspbian.

szpajder, 2017-10-15:
Docs clarified in release 2.4.0
Closing.

closed 2017-10-15 by szpajder