rtl-airband/RTLSDR-Airband #54: [Suggestion] display manual squelch level on waterfall

spotter-ssol, 2017-03-25:
Just a suggestion: with squelch set to auto, the second field on waterfall displays the current noise floor (AFAIK). When using manual squelch, that field always shows "100". I think it would be interesting to have the channels' manual squelch level in that field.

szpajder, 2017-05-18:
This is now implemented in unstable branch. Sorry for delay.

spotter-ssol, 2017-05-18:
Thanks! I'm eager to compile and test it. I'm also saw that there are some improvements in the code, like per channel squelch (AFAIK). Do we need to modify anything in the config file to use this feature?

szpajder, 2017-05-19:
Auto squelch per scanned channel is enabled by default. It is also possible
to set squelch manually per scanned channel, however this should hardly
ever be necessary in practice. There is an example of the new squelch
syntax in config/reference.conf. See also
https://github.com/szpajder/RTLSDR-Airband/issues/61 .

spotter-ssol, 2017-05-19:
Tested & working! Thanks to all of you!

closed 2017-05-19 by spotter-ssol