rtl-airband/RTLSDR-Airband #55: Compiler invocation, CC vs. CXX

philipz-59, 2017-04-06:
Hi,
 when cross compiling on my Ubuntu 16.04 x86_64 for armv7-generic it completely
ingnores CC variable settings, likely that C++ stuff is compiled with CXX.
 Would you consider either to add setting also for CXX variable or to specify
$(CC) compiler invocation, not to rely on default make behauvior?

Regrads!

szpajder, 2017-04-06:
.cpp files are compiled with $CXX which is set to g++ by make's default and
I see nothing wrong with this.

CC variable is set to g++ in the makefile, so that the final linking
process is done with g++ and not with cc (which is the default), otherwise
it will fail.

CFLAGS and CXXFLAGS are both set to the same value.

Do you need to switch to a different compiler binary or change the flags?
How do you invoke make, what commands do you expect it to generate and what
commands it generates instead?

philipz-59, 2017-04-06:
Yes, $CXX is set to g++ by default, but I need it to be armv7-linux-gnueabi-g++ for cross compiling.
When sitting on x86_64, it logically complains -march=armv7-a -mfpu=neon-vfpv4 unknown options.

Why you specify $CC which is not used, but not $CXX which is actually used?

I mean to make it more corosscompilation friendly, to save time elaborating with man pages.

szpajder, 2017-04-06:
>
> Yes, $CXX is set to g++ by default, but I need it to be
> armv7-linux-gnueabi-g++ for cross compiling.
> When sitting on x86_64, it logically complains -march=armv7-a
> -mfpu=neon-vfpv4 unknown options.
>

Then just modify the makefile - change the line:

    export CC = g++

to:

    export CC = $(CXX)

then run:

    CXX=armv7-linux-gnueabi-g++ PLATFORM=armv7-generic make

and let me know if this is sufficient.

Why you specify $CC which is not used, but not $CXX which is actually used?
>

Both $CXX and $CC are used. $CXX is left at its default, while $CC is not.

philipz-59, 2017-04-06:
Just now I am not solving how to compile. My point is to bring this valuable information
also to another future users. I mean to
1\ mention in documentation that $CXX setting is necessary for cross compilation, or
2\ preset $CXX in makefile

szpajder, 2017-04-06:
> Just now I am not solving how to compile. My point is to bring this valuable information
> also to another future users.

Great, but I want you to test the change I'm about to implement :-)

>  I mean to
> 1\ mention in documentation that $CXX setting is necessary for cross compilation, or

I don't think rtl_airband's README is the right place for a cross-compilation howto. This is rather advanced topic, I assume whoever attempts this is knowledgeable enough. If not, he/she can always compile the program on the target platform and this is what most people actually do.

> 2\ preset $CXX in makefile

I think it would be enough if the Makefile did not override $CXX or $CC if it's set by the user on the make command line, do you agree?

philipz-59, 2017-04-18:
My platform does not meet performance criteria (load >1.0). It is TI AM3352 Cortex-A8, about 2000 DMIPS. Consider it closed! Regards.

closed 2017-04-18 by szpajder