rtl-airband/RTLSDR-Airband #56: [Feature Request/Suggestion] Seperate File for each transmission

gnarpaws, 2017-04-06:
I'd like to be able to record transmissions into a new file for each transmission *instead* of having a new file every hour.
This is very important in my scenario because i need somewhat accurate timestamps.

**e.g.**
`<Filename Prefix>_20170406_17-21-20.mp3` (HH-MM-SS)
instead of
`<Filename Prefix>_20170406_17.mp3`

vaisala, 2017-04-28:
Other nice record file options would a single file and and new file every day.

bytepoet, 2017-05-26:
would love to see this with the corresponding frequency that it arrived on as well.

mdickw, 2017-10-26:
I accomplished this quick and dirty with a few modifications to output.cpp mostly. Obviously this would need to be implemented in a more robust way to make it into the code, but if you need to get individual files **right now**, you can try these modifications. This will give you a new file if there hasn't been anything in 10 seconds.

rtl_airband.h
modify struct channel_t {}
add
```
time_t start_time;
time_t last_time;
```

output.cpp
process_outputs() if (channel->outputs[k].type == O_FILE)
add/modify the following
```
if ((int)t - (int)channel->last_time > 10) {
	channel->start_time = t;
}

if(use_localtime)
	tmp = localtime(&channel->start_time);
else
	tmp = gmtime(&channel->start_time);
```

then modify per your requirements

`strftime(suffix, sizeof(suffix), "_%Y-%m-%d %H:%M:%S.mp3", tmp)`

at the end of the if, after channel->outputs[k].active = (channel->axcindicate != ' '), add

`channel->last_time = t; ()`

mdickw, 2018-02-08:
Going along with this, I've also found it to be very useful to track the signal level and start a new recording whenever that changes significantly. Helps split up the conversations a bit better.

Kaptronik, 2018-02-26:
Thank you @mdickw I will try this. Definately something I would like to see implemented in the main code.

metaphoricalmarlinspike, 2018-04-20:
Has anyone been using the code that @mdickw submitted? If so, how has it been working out? This is definitely a feature that I would like to see implemented.

gustavors01, 2018-07-20:
any news regarding this topic? It's allready implemented?

gustavors01, 2018-07-23:
Tks for reply.

gustavors01, 2018-07-27:
Please find commit https://github.com/gustavors01/RTLSDR-Airband/commit/514012677b443ea9c9d94f6b98244ee6da8b5166  for Split output mp3 file, per transmission, based on ScanOC @dreinhold commit. https://github.com/ScanOC/RTLSDR-Airband/commit/5faceddfa78645d14574d23e2a8fe5315d4f8527 .

On config file output need to use `split_trans = true`, file will have ...HHMMSS.mp3. 
Default is false and nothing changed, file name ...HH.mp3.

I am not a programmer than use with care, in my scenario it`s working fine.