rtl-airband/RTLSDR-Airband #57: Error compiling with x86 target on AMD Geode

vaisala, 2017-04-23:
The device uses an AMD Geode processor and has 512MB RAM, i intent to make a small low power receiver but cant get to compile for this x86 target.

Using PLATFORM x86

> 
> futro8 RTLSDR-Airband # make
> g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native   -c -o rtl_airband.o rtl_airband.cpp
> rtl_airband.cpp: In function ‘void demodulate()’:
> rtl_airband.cpp:411:104: warning: SSE vector return without SSE enabled changes the ABI [-Wpsabi]
>     __m128 a = _mm_set_ps(levels[*(buf2 + 3)], levels[*(buf2 + 2)], levels[*(buf2 + 1)], levels[*(buf2)]);
>                                                                                                         ^
> In file included from rtl_airband.cpp:37:0:
> /usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:944:1: error: inlining failed in call to always_inline ‘__m128 _mm_set_ps(float, float, float, float)’: target specific option mismatch
>  _mm_set_ps (const float __Z, const float __Y, const float __X, const float __W)
>  ^
> rtl_airband.cpp:411:104: error: called from here
>     __m128 a = _mm_set_ps(levels[*(buf2 + 3)], levels[*(buf2 + 2)], levels[*(buf2 + 1)], levels[*(buf2)]);
>                                                                                                         ^
> In file included from rtl_airband.cpp:37:0:
> /usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:922:1: error: inlining failed in call to always_inline ‘__m128 _mm_load_ps(const float*)’: target specific option mismatch
>  _mm_load_ps (float const *__P)
>  ^
> rtl_airband.cpp:412:41: error: called from here
>     __m128 b = _mm_load_ps(&window[i * 2]);
>                                          ^
> In file included from rtl_airband.cpp:37:0:
> /usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:193:1: error: inlining failed in call to always_inline ‘__m128 _mm_mul_ps(__m128, __m128)’: target specific option mismatch
>  _mm_mul_ps (__m128 __A, __m128 __B)
>  ^
> rtl_airband.cpp:413:24: error: called from here
>     a = _mm_mul_ps(a, b);
>                         ^
> In file included from rtl_airband.cpp:37:0:
> /usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:971:1: error: inlining failed in call to always_inline ‘void _mm_store_ps(float*, __m128)’: target specific option mismatch
>  _mm_store_ps (float *__P, __m128 __A)
>  ^
> rtl_airband.cpp:414:33: error: called from here
>     _mm_store_ps(&fftin[i][0], a);
>                                  ^
> In file included from rtl_airband.cpp:37:0:
> /usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:889:1: error: inlining failed in call to always_inline ‘__m128 _mm_set1_ps(float)’: target specific option mismatch
>  _mm_set1_ps (float __F)
>  ^
> rtl_airband.cpp:472:55: error: called from here
>      __m128 agccap = _mm_set1_ps(channel->agcmin * 4.5f);
>                                                        ^
> In file included from rtl_airband.cpp:37:0:
> /usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:929:1: error: inlining failed in call to always_inline ‘__m128 _mm_loadu_ps(const float*)’: target specific option mismatch
>  _mm_loadu_ps (float const *__P)
>  ^
> rtl_airband.cpp:474:49: error: called from here
>       __m128 t = _mm_loadu_ps(channel->wavein + j);
>                                                  ^
> In file included from rtl_airband.cpp:37:0:
> /usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:223:1: error: inlining failed in call to always_inline ‘__m128 _mm_min_ps(__m128, __m128)’: target specific option mismatch
>  _mm_min_ps (__m128 __A, __m128 __B)
>  ^
> rtl_airband.cpp:475:63: error: called from here
>       _mm_storeu_ps(channel->waveref + j, _mm_min_ps(t, agccap));
>                                                                ^
> In file included from rtl_airband.cpp:37:0:
> /usr/lib/gcc/i586-linux-gnu/4.9/include/xmmintrin.h:978:1: error: inlining failed in call to always_inline ‘void _mm_storeu_ps(float*, __m128)’: target specific option mismatch
>  _mm_storeu_ps (float *__P, __m128 __A)
>  ^
> rtl_airband.cpp:475:64: error: called from here
>       _mm_storeu_ps(channel->waveref + j, _mm_min_ps(t, agccap));
>                                                                 ^
> <builtin>: recipe for target 'rtl_airband.o' failed
> make: *** [rtl_airband.o] Error 1
> `
> 

This is a small Geode x86 machine (low power headless ) , so if i add -msse4.1 compile fine but will not run on my cpu.

> futro8 RTLSDR-Airband # cat /proc/cpuinfo
> processor       : 0
> vendor_id       : AuthenticAMD
> cpu family      : 5
> model           : 10
> model name      : Geode(TM) Integrated Processor by AMD PCS
> stepping        : 2
> microcode       : 0x8b
> cpu MHz         : 498.060
> cache size      : 128 KB
> fdiv_bug        : no
> f00f_bug        : no
> coma_bug        : no
> fpu             : yes
> fpu_exception   : yes
> cpuid level     : 1
> wp              : yes
> flags           : fpu de pse tsc msr cx8 sep pge cmov clflush mmx mmxext 3dnowext 3dnow vmmcall
> bogomips        : 996.12
> clflush size    : 32
> cache_alignment : 32
> address sizes   : 32 bits physical, 32 bits virtual
> power management:

szpajder, 2017-04-24:
> flags : fpu de pse tsc msr cx8 sep pge cmov clflush mmx mmxext 3dnowext
> 3dnow vmmcall

There is no SSE support it this CPU, hence the program will not work on it.

vaisala, 2017-04-24:
I understand that this is a fairly old computer, but for the task of streaming one or two airband channels should be more than enough. I run right now your excellent vdl2 decoder on this same computer with a 30% CPU load ,  very nice for a small silent (fanless) computer.

Some CPUs were released after SSE2 was developed, but did not implement it AMD CPUs prior to Athlon 64, such as Athlon XP, Geode, VIA C3 or Transmeta Crusoe. 

It would be possible to have a "generic Intel x86" same way as generic arm builds are provided to make use of those old but still perfectly functional machines?

szpajder, 2017-04-24:
My Raspberry Pi v1 with 700 MHz CPU is happily running dumpvdl2 with a
single VDL2 channel. It takes about 25-30% of CPU, which is nearly the same
as your 500 MHz Geode. The CPU usage increase is linear (approximately)
with the number of configured channels.

RTLSDR-Airband is a whole different thing. No matter how many channels you
are streaming, the whole 2.56 MHz of bandwidth is analyzed with FFT
channelizer and this is where most of the computing power is spent on.
Adding a channel or two would not make a huge difference - your 500 MHz CPU
would already be saturated with FFT calculation. This is what happens on
the Pi when running rtl_airband with GPU acceleration disabled. CPU is
constantly at 100% and the audio chokes heavily even with single channel.

It's not that rtl_airband's code is so terribly inefficient. The program is
designed to stream multiple channels. For one or two channels it is more
efficient to do direct demodulation instead of FFT channelization, so you
could, for example, use rtl_fm for this purpose. However at 4 channels or
above FFT clearly wins with direct demod, because the CPU usage of the
latter algorithm increases linearly with number of channels, while in FFT
it's more or less constant.

While it is technically possible to modify rtl_airband's code, so that it
could run on a non-SSE CPU, it is quite pointless, because if the CPU has
no SSE it's probably very outdated and not beefy enough to handle the task.
Keep in mind that armv7-generic and armv8-generic are actually using
vectorized FPU instructions which work similarly to SSE. Just GPU
acceleration is disabled on these builds, but vectorization is still there
- otherwise these CPUs would not be able to cope with the workload.

vaisala, 2017-04-25:
Thanks for taking the time to write the details. Time to look for a new small form factor PC.

closed 2017-04-25 by vaisala