rtl-airband/RTLSDR-Airband #60: Wish: Make a SoapySDR version of Airband

starvald, 2017-05-08:
SDR's like Airspy and SDRPlay offer better sensitivity and dynamic range but command line linux apps are pretty much unusable.
just unplugged my airspy r2 in the attic and put my rtlsdr v3 back on so i could use airband again.

wish i could use my airspy or sdrplay with airband.

szpajder, 2017-05-13:
Yup, sounds like a good idea. I'm not yet sure if it will be possible to keep all RPi-related optimizations with SoapySDR in place, but will investigate this topic one day.

starvald, 2017-05-13:
Thanks for thinking about it.

szpajder, 2018-01-03:
SoapySDR is supported in unstable branch.

Docs are not yet updated, so here is a quick howto:
```
devices:
({
  type = "soapysdr";
  device_string = "driver=rtlsdr,rtl=0";
  gain = 30;
  centerfreq = 150000000;
  correction = 77;
  channels:
  (
   ....
  )
});
```
device_string uses the same format as `--find` or `--probe` args of SoapySDRUtil. rtl_airband does not care what you put in there - if it's syntactically correct and the driver understands it, it should work. A more fancy example utilizing SoapyRemote to read sample data from a dongle connected to a different machine:
```
device_string = "remote=10.0.0.5,driver=remote,remote:timeout=100000,remote:driver=rtlsdr,rtl=0"
```
Sampling rate can be set with `sample_rate` option in the device section (eg. `sample_rate = 8000000` for 8 Msps).

starvald, 2018-01-04:
awesome look forward to trying it out at the weekend.

starvald, 2018-01-05:
Seems to be working fine with my sdrplay rsp2, took a little figuring out i had to add WITH_SOAPYSDR=1 to get soapy support working.

My Airspy R2 i'm having an issue with frequency is shifted, 156MHz signal is in 156.375MHz and the recording was a quarter speed, this is the same problem i had with rx_fm part of rx_tools a soapy based rtl_fm port.

but as long as it's working in sdrplay i'm happy.

szpajder, 2018-01-05:
Most probably the sampling rate of the device is different than the program
thinks it is.

I've skimmed through libairspy code and it looks like it only supports
2500000 and 10000000 sps rates, correct? The default sample rate in
rtl_airband is 2560000 - if you get the recording at quarter speed, it
probably means the device is running at 10 Msps (and because it is not
exactly four times more, you get a freq offset as a bonus). Try to add this
setting to the device section in rtl_airband.conf (at the same level where
you set type, device_string and centerfreq):

    sample_rate = 10000000;

or this:

    sample_rate = 2500000;

and check out if it cures the problem.

starvald, 2018-01-05:
already using  sample_rate = 10000000 but will try with 2500000.

szpajder, 2018-01-07:
I've pushed an update to soapysdr code, it now adjusts the default sampling rate of 2560000 to the nearest rate supported by the device (as returned by `SoapySDRDevice_getSampleRateRange()`). On Airspy it should now automatically set 2500000, so the quarter-speed problem should now be fixed.

If the `sample_rate` knob has been set in the config, then it's used directly, without checking if the HW supports it or not. I assume that the user knows, what he's doing.

starvald, 2018-01-08:
Have it running for a few days now, working great on my rsp2.

Adraenyse, 2018-01-08:
This is great news! Are we still limited to 8 frequencies when using the soapy interface though?

szpajder, 2018-01-08:
Number of channels per receiver is now unlimited (regardless of HW
interface).

szpajder, 2018-01-23:
A few soapy-related improvements in unstable branch:

- Fix for improper handling of sample rates which are not an exact multiple of output audio rate. This caused garbled audio, eg. on Airspy Mini with sample_rate = 3 Msps.

- Configurable SoapySDR channel number (only relevant for SDRs supporting more than one channel) - example: `channel = 1` (default: 0).

- Configurable antenna (`antenna = "antenna name";`)

- Per-component gain setting - relevant for SDRs with several gain components. Example for Airspy: `gain = "LNA=15,MIX=15,VGA=5.5";`. `gain` option can still be given the old way (ie. as a single number) - SoapySDR driver will then attempt to balance the gain setting on individual components to achieve the configured value. A list of supported gain element names and their ranges can be obtained with this command: `SoapySDRUtil --probe=driver=<driver_name>`.

szpajder, 2018-02-10:
Released in version 3.0.0.

Closing.

closed 2018-02-10 by szpajder