rtl-airband/RTLSDR-Airband #61: agcmin should be kept on a per-frequency basis; add per-frequency squelch

strix-technica, 2017-05-09:
My typical configuration is to scan about 5 frequencies, but the noise floor varies between each frequency significantly.  rtl_airband auto squelch performs poorly in these circumstances because it appears that it `channel->agcmin` averages the noise across all frequencies rather than maintaining separate data for each frequency.

As a result, I've had to set the squelch manually but since there is only one global squelch setting, there is no one setting that is good for all frequencies.  Being able to set squelch on a per-frequency basis would be excellent.  Even better, maintain separate `agcmin` per frequency so autosquelch has a chance as well.

I would imagine that this would be almost as applicable to multi-channel mode as scanning mode.

If you don't have time for this, I could potentially have a go at it and, even though my C's a bit rusty, it doesn't look too terribly difficult to implement.  Would you be interested on a PR and/or working together on this?

szpajder, 2017-05-13:
The current scanning mode implementation was created as a quick hack, just to check if it would work at all. The demodulator has no information about which channel/frequency the currently processed sample block was received on. There are several feature requests related to scan mode, which are very hard to implement now (see, for example https://github.com/szpajder/RTLSDR-Airband/issues/59, https://github.com/szpajder/RTLSDR-Airband/issues/39), so I think that instead of further hacking the current scanning code, it's better to reimplement it in a more flexible and clean way. I need some time to make up my mind of how this should work.

strix-technica, 2017-05-13:
> The demodulator has no information about which channel/frequency the currently processed sample block was received on.

Sure.  I can see `rtl_airband.cpp:controller_thread()` cycle through the frequencies, updating only the centre frequency.  Nothing else knows of the change.

> There are several feature requests related to scan mode, which are very hard to implement now (see, for example #59, #39)

Yes, I saw those.  As you say, some of them are not straight forward to implement.

>so I think that instead of further hacking the current scanning code, it's better to reimplement it in a more flexible and clean way. I need some time to make up my mind of how this should work.

Doubtless better, but a major refactoring would also likely push out significantly this proposed modification, especially if how best to refactor is not yet clear.

I *think* the modifications necessary to implement this enhancement request should be relatively straight forward; here's what I had in mind:
* make `struct channel_t` members `agcavgfast, agcavgslow, agcmin` and `sqlevel` pointers same as `freqlist` and `labels`;
* add member `int freq_idx` to indicate which is the currently selected frequency;
* for each of those members, update references accordingly (eg `channel->agcmin` becomes `channel->agcmin[channel->freq_idx]`);
* to `rtl_airband.cpp:controller_thread()` add `dev->channels[0].freq_idx = i;`
* update `config.cpp:parse_channels()` accordingly to initialise arrays rather than singletons; and
* adapt same to cope with squelch being a singleton or array along with any other necessary modifications to `config.cpp` to preserve backward compatibility for existing config files.

I have no idea what sort of refactoring you had in mind but I don't imagine that this coincides well with it.  That being the case, I can fork and hack for my own purposes; the only question is whether I can do so in a way that better fits your future plans and thus be of more use to you.

Since those future plans are not yet clear, I'll go ahead and try out those changes when I get a chance.  I'll send a PR if you like and if you like it you can use it.  If you hate it, it can always stay as it is until you've had a chance to think it over ;)

strix-technica, 2017-05-15:
Okay, that didn't take long and early indications are that results in scanning mode are substantially improved — so much so that I no longer *need* to manually set a squelch.  (Setting squelch per frequency does improve matters a little, but it seems to work fairly well without.)  

Multi-channel mode seems to work, to the extent that I've been able to test it (only a single channel defined in multi-channel mode).

I'll test a while and then send a couple of PRs.  There are some changes I think you should make anyway whether you accept my squelch mod or not, hence two separate PRs.

szpajder, 2017-10-15:
Released in v2.4.0.
Closing.

closed 2017-10-15 by szpajder