rtl-airband/RTLSDR-Airband #65: 8.33 conversion

tnadolecki, 2017-07-01:
Can you pls include the table and auto-detection mechanism?

Thanks

http://g3asr.co.uk/calculators/833kHz.htm

szpajder, 2017-07-02:
Is there any particular problem which you expect to be fixed by this feature?

Note that tuning precision in rtl_airband is limited to a couple of kHz by the resolution of FFT channelizer. So whatever you enter at the fourth decimal place is mostly irrelevant.

tnadolecki, 2017-07-02:
No problem at all. This is not an issue - It would be rather nice feature to have since lot of conversion is going on at the moment.

szpajder, 2017-07-03:
But do you actually need to do this conversion?

If you just put the unconverted frequency into your config, it means your
tuning will be off by 5 kHz maximum, so it should still work correctly. If
not - maybe you could try enabling AFC for this channel and check if it is
able to correct the offset automatically?

tnadolecki, 2017-07-03:
Hi,
I don't need it myself, On top of converted freqs I'm also utilizing AFC on every channel, it does excellent job, I though this might be nice feature to have and it might be useful for others if software could automatically adjust freq and indicate in foreground mode.
Technically there is no need for it as long as afc can adjust up to 5khz then all should be fine.
Sorry for raising an issue I should have posted it somewhere else perhaps as feature reuqest.
I love the software btw. Great job.
Thanks. Tom.

closed 2017-07-03 by tnadolecki