rtl-airband/RTLSDR-Airband #66: Choppy audio running on 2 dongles

spotter-ssol, 2017-07-06:
I was sucessfuly running rtl_airband (1 rtlsdr v3 dongle) and dump1090 (1 fa pro dongle) over a raspberry pi 2 (25W psu) for + a year. There's also a flashdrive plugged to store the audio file output. Yesterday I've inserted a third dongle to expand my airband audio coverage. All of them plugged directly without external hub, with max_usb_current=1 set on /boot/config.txt.
Now the audio output became erratic. There's aways some cut or silence inserted between. If there's an audio message, let's say, "ABCDEFGH", the output goes like "ABCFGH", or "ABC...FG.", or "ACD.FG.". This affect both streaming and file output. In my setup, both dongles are configured for multichannel, with just 2 channels for the first and 3 for the second.
Tried lowering rtlsdr_buffers up to 5, no effect.
Tried raising GPU memory up to 256MB, no effect.
Finally, stopped dump1090 and disabled file output to flash drive to let just the 2 airband dongles streaming, but aso no effect.
Am I missing something?

szpajder, 2017-07-06:
I have an identical setup and I'm having the same issue. RPi 2 is not beefy
enough for this task. I'm about to upgrade to RPi3 in a couple of days to
solve this.

spotter-ssol, 2017-07-09:
Today I've switched my device to a RPi3 and looks like the problem is gone. Thanks for the tip!

closed 2017-07-09 by spotter-ssol