rtl-airband/RTLSDR-Airband #67: Distorted NFM recordings.

starvald, 2017-07-07:
I monitor marine band and sometimes the recordings are distorted, you can just make out what they say most of the time but it is annoying,

zipped example
[CH13_20170701_08.zip](https://github.com/szpajder/RTLSDR-Airband/files/1131105/CH13_20170701_08.zip)

any ideas?

lzieg, 2017-07-07:
I have the same problem on the aircraft band. In my case, it is quite clearly related to very strong signals. I am about 500 meters from the runway at our local airport, and some airplanes near my house come through distorted, just as you describe.

I assumed this caused by overload in the RTLSDR dongle. I improved the situation by reducing the gain on the RTLSDR to 12. Some close aircraft still sound distorted, but the problem is reduced. The drawback is, of course, less sensitivity to pick up weaker aircraft, so the gain of 12 is a compromise between distortion on close aircraft and sensitivity to pick up distant aircraft.

I'm not sure if the problem can be solved in rtlsdr-airband if it is indeed due to limited dynamic range in the hardware, but it would be nice if something could be done.

Lee

starvald, 2017-07-07:
yeah problem with the marine band is the coast guard signal is very strong and boats out to sea weak.

szpajder, 2017-07-07:
If you suspect that these distortions are due to clipping of the input signal, you may verify this easily just by lowering the gain. If the problem goes away, that was the cause. In this case there is not a whole lot you could do about this, apart from keeping the gain reasonably low. AM is more problematic in this case, because the information is encoded in amplitude, so lowering the gain will indeed reduce your ability to pick up distant transmissions. However in FM the information is encoded by changes of instantaneous frequency which makes it more immune to this problem.

Such distortions may also be caused by improper PPM correction setting. Make sure you have calibrated your dongle correctly and put the correct value in your config. See https://github.com/szpajder/dumpvdl2/blob/master/README.md for some hints on how to do the calibration (the relevant FAQ section is "How do I estimate PPM correction value for my dongle?")

Also try listening to this particular channel using some other program (eg. SDRSharp or GQRX). At least you would know if the problem is due to the input signal quality or due to rtl_airband's config.

starvald, 2017-07-07:
i'll try turning down the gain, using an rtlsdr blog v3 so ppm is pretty stable for me at -1.

tnadolecki, 2017-07-07:
Turning down gain to remove clipping only works for AM or SSB . 
In FM try with 5kHz bandwith first, then go lower if required, may walkie-talkies are set to 2.5kHz modulation. For NOAA satellites use 17kHz,

starvald, 2017-07-08:
it is too much gain, testing with CubicSDR.
now to see how much gain is optimal.

pulcak, 2017-07-13:
Hi, the same problem by me. I am using szpajder and best RTLSDR dongles (temp. Compensated) to stream AM ATC traffic. It seams that szpajder does not use AGC or is it very poor.  Near transmissions are distorted, of course, because I don't want to lose distant ones by lowering RF gain. Finding optimal setting for airplanes in the neighborhood and distant ones is impossible. 
It is somehow clear to me that if I am using more than one channel from each DVB-T stick the AGC on one channel would cause amplification/dampening on the other. 

Could there be an option to turn on (strong) AGC ? Simultaneous transmissions over different channels on the same stick are very rare, so I would not lose much if AGC on first channel would damp receive on the second one. I don't know if AGC on per channel basis is possible but at least one AGC on all 2.5mhz bandwidth?

szpajder, 2017-07-13:
R820T tuner has a hardware AGC feature, however it only works correctly for wideband signals (eg. DVB-T). For narrowband AM channels it usually sets the gain way too high. HW Auto gain was supported in RTLSDR-Airband version 2.0.0, but was later removed because people often complained about audio bleeding between channels, which was caused by excessive gain. Just download version 2.0.0 and check it out by yourself, if you wish.

As far as software AGC is concerned, this is a no-go because on Raspberry Pi samples are processed in large batches for effective GPU usage. This causes some processing delay. If a powerful transmission from a nearby aircraft pops up, it's already too late to tweak the gain to avoid receiver overdrive, because some distorted samples have already been processed. Setting the dongle gain also takes a bit of time, it's not an instant operation. In the end, you would still get some distorted audio at the start of the transmission, which would sound awful. I think the right way to go is to add support for SDR hardware with more dynamic range (like SDRPlay or Airspy). I prefer to spend my time and resources on this rather than on crude AGC hacks which would not improve things a lot.

If you need different gain settings for two adjacent channels, then my tip for you is: buy yourself a second dongle and switch one of the channels to it. For a mere 15 bucks you will get a 100% HW-based solution which would easily outperform any kind of SW-AGC algorithm which anybody could come up with.

pulcak, 2017-07-13:
Thanks for very quick and good answer. What I was thinking is the feature you are mentioning in the second part of your answer. What I think is that if the first 100ms or so of the transmission (delay for decision and sending command to dongle) is distorted I think it is far better the the whole one. You could also delete this part. This would i call "slow AGC". Something like that would very improve my streaming. Such thing can't be done in any other way in my knowledge. 

For me there is no problem using one dongle per channel. I am already using three dongles using one to four channels per dongle using different gain setting hence with the same issue. Luckily this huge problem persists only on the channel where planes are flying over my antenna and landing 10km away, where it is very difficult to hear them on the ground. 

Anyway if you programmed this thing on RPI I must say that such program is far better than everything I have ever seen on such platform. I'm also very enjoying mixer feature. Congratulations and I still hope to get some improvements...

Regards, 
Peter

spotter-ssol, 2017-08-15:
> I am already using three dongles using one to four channels per dongle using different gain setting hence with the same issue.

Sorry for the offtopic, but @pulcak, would you mind to share your hardware/software setup?

closed 2017-09-24 by szpajder

gustavors01, 2018-07-20:
@starvald Dr. Mr. Lee Donaghy, you like to share your project or comments for better setup?
I like to start a similar project recording around 8 marine channels.

starvald, 2018-07-21:
@gustavors01 this is my current setup for an Airspy R2, records to mp3 and plays over pulseaudio.

[](https://paste.ubuntu.com/p/2TjzBDcymJ/)

gustavors01, 2018-07-23:
Many tks for Sharing.