rtl-airband/RTLSDR-Airband #68: How to compile for windows ?

Khoosham, 2017-07-18:
Hello
is there any tutorial to compile it for windows or maybe u can compile it and include the executable on this repo itself so that users only have their own configuration to make.
Thanks in advance :)

szpajder, 2017-07-18:
Windows support has been made by the original program author, but later it
was unmaintained for a long time. It no longer worked with the new
features, so it has been dropped.

Khoosham, 2017-07-18:
yes i saw it on microtony repo... its outdated. no problem
i only have one issue i compiled it for raspberrypi
the only problem i have if i connect my dongle ..reboot my pi and start the rtlsdr for liveatc

i run rtltest -t and get this error USB Claim Interface Error 6
if i disconnect and connect dongle again then it works 

idk how to solve that error i look everywhere on google maybe u can help me

Thanks man :)

szpajder, 2017-07-18:
You start rtl_test while rtl_airband is running? Then it's quite normal -
it can't access the dongle and errors out.

Only one app can access the dongle at a time.

Khoosham, 2017-07-18:
well i mean i don't run both at the same time only 1
if i run rtltest - t i get this error 
then obviously rtlairband wont work as i get the above error on rtltest

i tried everything blacklist driver and still not work

szpajder, 2017-07-18:
Probably some process which uses the dongle is being started automatically during boot. When you unplug the dongle, it terminates - that's why you can use the dongle with other apps then.

Run `lsusb`, find your dongle in its output, for example:
```
Bus 001 Device 008: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
```
There should be a device node `/dev/bus/usb/<bus_number>/<device_number>`, ie. `/dev/bus/usb/001/008` in this case. Check who is using it:
```
root@rasp03:~# lsof /dev/bus/usb/001/008
COMMAND   PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
rtl_airba 758 root   15u   CHR  189,7      0t0 7425 /dev/bus/usb/001/008
```
There you got it - it's the rtl_airband process, PID 758. Once you kill it, you will be able to access the device.

(`apt-get install lsof` if you don't have it installed).

Khoosham, 2017-07-19:
Thanks a lot it worked i have follow ur instructions again to install rtlsdr and airband now at startup usb is not used by the process 👍

closed 2017-07-19 by Khoosham