rtl-airband/RTLSDR-Airband #70: Config Error: unknown modulation

USAFPride, 2017-07-31:
Running the unstable version (thanks for the "serial" add).  Added 2nd dongle to rtl_airband to pick up the NFM noaa weather radio.  With the attached configuration, I get the following error: 
`Configuration error: devices.[1] channels.[0]: unknown modulation`

I must be missing something simple, but I'm stumped.

[rtl_airband.conf.txt](https://github.com/szpajder/RTLSDR-Airband/files/1186125/rtl_airband.conf.txt)

spotter-ssol, 2017-07-31:
... 
channels:
  (
      {
        modulation = "am";
        freqs = ( 119300000, 121800000, 135650000, 124600000, 128400000 );
...

USAFPride, 2017-07-31:
@spotter-ssol, `modulation = "am";` is the default state.  I added it and it still produces the same error.

szpajder, 2017-07-31:
make clean
make PLATFORM=<your_platform> NFM=1

should do the trick.

USAFPride, 2017-07-31:
@szpajder - simple RTFM issue.

closed 2017-07-31 by USAFPride