rtl-airband/RTLSDR-Airband #71: High CPU usage on FreeBSD

nyammy, 2017-09-08:
I'm not sure you're interested in FreeBSD but if you have any idea or workaround to solve this, I hope you to enhance/add some codes for FreeBSD.

I compiled rtl_airband on FreeBSD 10.3-Release and it's working, demodulating correctly, saving MP3 file, and streaming to my icecast server.
But it uses 100% power of a CPU core like this:

-----
CPU: 36.3% user,  1.1% nice,  1.8% system,  0.2% interrupt, 60.7% idle
(This CPU has 4 cores so it's eating 25% of total CPU power)

  PID USERNAME    THR PRI NICE   SIZE    RES STATE   C   TIME    WCPU COMMAND
 4977 root          5  20    0 79200K 15380K nanslp  2  42.9H 100.78% rtl_airband

-----

STATE is nanslp = nanosleep so it should spend most of time sleeping, but eating CPU power.

Compile could be done like below.

Installed necessary libraries like:

    # pkg install rtl-sdr lame fftw3 fftw3-float libshout libvorbis libconfig
    # git clone https://github.com/szpajder/RTLSDR-Airband.git
    # cd RTLSDR-Airband/

Edited x86 part of makefile like:

    CFLAGS += -march=native -I/usr/local/include
    LDLIBS += -lfftw3f -lc++
(and change all "-g root" to "-g wheel" in install part)

And continued like:

    # PLATFORM=x86 gmake
    # gmake install

pid file should be located in /var/run so change rtl_airband.conf to use this place, not /run.

Example of startup script for rc.d:

    #!/bin/sh
    
    # PROVIDE: rtl_airband
    # REQUIRE: DAEMON
    # BEFORE: LOGIN
    # KEYWORD: nojail shutdown
    
    . /etc/rc.subr
     
    name=rtl_airband
    rcvar=rtl_airband_enable
    
    command="/usr/local/bin/rtl_airband"
    
    load_rc_config ${name}
    run_rc_command "$1"

Thanks,

szpajder, 2017-09-08:
Can you do any kind of profiling to find out where this nanosleep gets
invoked? There are at least a couple of places and I don't have any BSD at
hand to troubleshoot this.

nyammy, 2017-09-08:
Now I know only that output_thread is using over 60% of the time.
Here is the output of "hotuser" script (utility tools works with dtrace):

    FUNCTION                                                COUNT   PCNT
    ...
    libthr.so.3`0x801175669                                   358   1.9%
    libthr.so.3`0x801175671                                   411   2.2%
    libthr.so.3`0x801175679                                   415   2.2%
    libthr.so.3`__pthread_cond_wait                           832   4.4%
    rtl_airband`output_thread(void*)                        12616  67.2%

I'll continue to probe later.

nyammy, 2017-09-10:
OK, I found the solution. The problem is in mutex handling.
Mutex must be locked before calling, but in your code, mp3_mutex is not locked when first pthread_cond_wait() is called in output_thread.
pthread_cond_wait() locks mutex just before wake-up so it will be correct condition if first wait-wake process successfully completed, but it fails on some environment.
So I've added

    pthread_mutex_lock(&mp3_mutex);

just before 

    while (!do_exit)

in output_thread.
Now it's using ~3% power of a core running at 1.6GHz. Significantly decreased compared to before: 100% at 3.1GHz (automatically boosted to max speed).

Thanks.

szpajder, 2017-09-10:
I prefer to do this slightly differently (see https://github.com/szpajder/RTLSDR-Airband/commit/91671aee379d9db3f7ba1f3bd3a7920e03a301e9). Could you please pull the latest unstable branch and verify if that's OK for you?

nyammy, 2017-09-11:
OK, it's safer and It works. Unlocking just after wait means there are no conditions nor data which should be handled in locked situation, right?

(All of your macros are named in uppercase, but safe_cond_* are in lowercase. Is this OK?)

Also, could you add "x86_freebsd" or something like that to PLATFORM in makefile if possible?
I (We?) will be happy if it can be compiled without manual changes.

szpajder, 2017-09-12:
Yes, we don't have any shared resources here, the cond_signal is there only to trigger output handler to do its work.
I added x86-freebsd build target and your startup script. Pushed into unstable.

nyammy, 2017-09-13:
Great. Works fine.
Thank you!

szpajder, 2017-10-15:
Released in v2.4.0.
Closing.

closed 2017-10-15 by szpajder