rtl-airband/RTLSDR-Airband #73: automatic gain control ?

deltaflyer4747, 2017-09-16:
You state in readme.md that there is AGC available, however i do not see any doccumentation about that nor did i find any way to do this; only setting related to gain i see is static gain setting, which is very problematic for me :(

szpajder, 2017-09-16:
More appropriate name for this feature would be something like Automatic Volume Equalization. It works for AM only and it's done at post-demod stage to boost the audio volume for distant transmissions. It is always enabled and not configurable.

There is no AGC for RF signal. Once upon a time there was an option to turn on hardware AGC on R820T tuner, however it's totally unsuitable for narrowband channels and caused more harm than good. Hence, this feature has been removed in version 2.2.0.

deltaflyer4747, 2017-09-16:
Well, that is a problem, since most of the time the signal is quite weak but when airplane transmits closeby, it overloads Rx demod and you can't hear ... well, nothing.

szpajder, 2017-09-16:
It's due to low dynamic range of the RTL chip. Once we will support more capable SDR receivers in rtl_airband, this problem could be solved by replacing the RTL dongle with a better receiver.

deltaflyer4747, 2017-09-16:
Better SDR means better selectivity, better dynamic range, better A/D... and we are at price 3x higher - at best :) But hey, find me one that can do better (in airband) than 820T2 and i'll take it.

szpajder, 2017-10-15:
Release 2.4.0 includes clarified documentation.
Closing.

closed 2017-10-15 by szpajder