rtl-airband/RTLSDR-Airband #75: Channel frequencies

mdickw, 2017-10-26:
I'm currently receiving 8 NFM channels. Some work quite well, some not so much. The ones that I have trouble with seem to be fine when I use GQRX to decode them discretely. I'm wondering if those particular channels just don't quite align with the FFT channels.

How do I determine the frequencies of the FFT channels?

szpajder, 2017-10-27:
Bins are spaced every 5 kHz counting from centerfreq (upwards and
downwards).

szpajder, 2018-01-23:
The current code in unstable branch (https://github.com/szpajder/RTLSDR-Airband/tree/unstable) does an extra step to alleviate this problem - it now shifts the configured channel frequency to the center of the chosen FFT bin before performing NFM demodulation. If your problem is related to channel misalignment, it may help, so check it out.

And just to be more precise with my original answer to your question about FFT channel frequencies - rtl_airband uses even FFT bin alignment, ie. the center frequency is located at the edge of two adjacent bins and there are 256 bins to the left of it and 256 bins to the right. Which means if the receiver center frequency is, say, 150 MHz, then the center frequencies of the channels are: 150.0025, 150.0075, 150.0125, 150.0175 and so on. Same thing when going downwards from center frequency.

By the way, sampling rate and FFT size are now configurable, so the FFT bin width is no longer fixed at 5 kHz - it's equal to `sample_rate/fft_size`.

starvald, 2018-01-26:
@szpajder how you you change the FFT size?
i get what sounds like overloading on my sdrplay but i think that's due to old marine channels being 25KHz, tested reducing channel bandwidth on cubicsdr and get the same effect.

szpajder, 2018-01-26:
At top level (ie. outside of `devices` section):
```
fft_size = 1024;
```
Supported values: powers of two between 256 and 8192. This setting is global for all devices. The default is 512.

Having the channel bandwidth set too wide usually does not have adverse effect on sound quality - background noise may be a little more audible, but that's all. On the other hand, setting this too narrow will clip the channel and the audio might be garbled. So don't be tempted to drive `fft_size` too high. I would rather play with gain settings and see if it makes a difference. What's your current gain setting for SDRPlay in rtl_airband? Are you running the latest unstable code?

starvald, 2018-01-26:
was using 
 gain = 36
just changed to
 gain = "IFGR=36,RFGR=4";

which are the same setting i've used in Cubic with no distortion.
would RFGR change with 'gain = 36'

yes now using the latest code.

szpajder, 2018-01-26:
Yes, gain = 36 is different than gain = "IFGR=36,RFGR=4".

First of all, SDRPlay does not have gain knobs but gain reduction knobs (hence the "GR" in their names). They work like attenuators. For RSP2, RFGR range is from 0 to 8. Setting it to 0 causes the built-in amplifier (LNA) to be set to its highest gain setting. RFGR=4 reduces LNA gain to -24dB from the max. IFGR range is from 20dB to 59dB. Again, highest gain is 20, lowest gain is 59.

When you set "gain = 36;" in rtl_airband.conf, you tell the SoapySDR: "please distribute the gain reduction across all available gain components automatically, so that end-to-end gain reduction is equal to 36". SoapySDR attempts to apply as much gain reduction on the digital side, as possible, while keeping the analog gain (ie. the LNA) high. In short: gain=36 is equal to IFGR=36,RFGR=0, which means the LNA is at its highest gain setting. Indeed this might give you an overload.

This is why I added the possibility of setting individual gain components manually. Automation might not always work as expected.

starvald, 2018-02-03:
changing the fft has worked for me, the distortions have gone from my sdrplay recordings.

mdickw, 2018-02-04:
I finally got a chance to check this out. I have a number of custom modifications that I need to stitch in whenever updating so it's a bit of a process.

Anyway, I ran it with fft size 1024 and there's a marked reduction in background static on the channel I was having trouble with. Some of my original troubles with that channel with were gain-related and I improved that a while back, but this gets reception quality to 99%.

The signal levels coming from airband are much lower (with no other settings changed besides updating and fft size).

closed 2018-02-10 by szpajder