rtl-airband/RTLSDR-Airband #76: Icecast Stream Description

USAFPride, 2018-01-01:
I'm attempting to get the "Stream Description" to show on Icecast servers v2.4.x.
Here is my current config (I've tried with `send_scan_freq_tags` both `true` and `false` and the description still will not show. Other streams are showing a "stream description", but I do not know the source of those streams.

```
 channels:
  (
      {
        modulation = "nfm";
        freqs = (162550000);
        labels = ("rPi3, RTL-SDR dongle, running RTLSDR-Airband");
        send_scan_freq_tags = true;
        squelch = 50;
        outputs: (
.......
```

szpajder, 2018-01-01:
Most probably you want to set `name` in the icecast output section?

Here is an example: https://github.com/szpajder/RTLSDR-Airband/blob/master/config/basic_multichannel.conf

USAFPride, 2018-01-01:
The `name` set in the `outputs` section is coming across as "Stream Name" on the server.  The "Stream Description" is not showing.  I've tried moving the `labels` and `send_scan_freq_tags` inside the `outputs` section and it still didn't show.

szpajder, 2018-01-01:
Nope, it's just a separate stream parameter which was not configurable in rtl_airband.

This is now implemented in unstable branch. Please pull and add `description` option to the icecast output section.

USAFPride, 2018-01-01:
Thanks.  How stable is the unstable version at this point?

szpajder, 2018-01-01:
It's pretty close to a stable release. Go for it.

USAFPride, 2018-01-01:
I just pulled a `diff` for patch `3d16324dbd3ce758c2be5d8f13d5e43f724c1ff1` and applied.  Works as expected.

closed 2018-01-01 by USAFPride