rtl-airband/RTLSDR-Airband #78: "Failed to submit transfer 7!"

mdickw, 2018-01-03:
Recently added a fifth dongle and I get "Failed to submit transfer 7!" when I start airband. If the fifth dongle is the only one in the airband config, airband just dies. If it's in there with another dongle, then it runs, but doesn't get anything from that dongle (still generates the error).

szpajder, 2018-01-03:
HW platform is...?

mdickw, 2018-01-03:
All RTL dongles.

szpajder, 2018-01-03:
Great, but what's the machine you plug them into?

mdickw, 2018-01-03:
Oh, right. It's x86. Ubuntu 16.04.

szpajder, 2018-01-03:
USB bandwidth problems? Make sure the dongles are not hanging off a single
USB channel (in particular they should not be connected to a common hub).

Do a simultaneous test of all receivers with "rtl_test -s 2560000 -d
<device_num> -t" - make sure that you are not losing samples or have
transfer submit failures. Sometimes it might be a matter of a particular
combination of USB hardware, kernel version and libusb version - some work
better, some worse.

mdickw, 2018-01-04:
Doesn't seem to matter how I move them around USB-wise or which ones I use. It just maxes out at four. Did find this in the reference config, though:

```
# Number of USB read buffers to be used per dongle
# The default value is 10. If you use four or more dongles simultaneously
# or you get error messages like "Failed to submit transfer 12!"
# you may need to lower this value
# rtlsdr_buffers = 10;
```

I set the buffers to 5 and I've got 5 dongles going okay (3 in use by airband, 2 in use by other programs using unknown number of buffers). That seems to have taken care of things.

closed 2018-01-04 by mdickw

mdickw, 2018-02-05:
On the latest unstable release, I'm having this issue again, even with rtlsdr_buffers set to 5. I wonder if anything has changed that might affect this?

szpajder, 2018-02-05:
Yes, the config option is now named "buffers" and it shall be placed inside
the relevant device section (ie. it now works on a per-device basis).

rtlsdr_buffers global option does nothing now.

I've started rewriting the docs. The new version is not yet complete, but
the device configuration section mostly is. See here:
https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-RTLSDR-devices

mdickw, 2018-02-05:
Awesome, thanks, that did the trick.