rtl-airband/RTLSDR-Airband #79: Feedback

nzstevem, 2018-01-21:
Hi, Great work. Just successfully built release 2,4 on the latest ubuntu release (17.10) running on an HP SFF x86/64. All good!

This link https://help.ubuntu.com/community/CompilingEasyHowTo has a few useful tips for those not that familiar with Linux that get you from a clean new O/S install to the point where your instructions kick in.  

Rgds Steve

szpajder, 2018-01-23:
Thanks for feedback. One day I'll probably switch to cmake or autotools to
make the build process a little less clumsy than it is now.

szpajder, 2018-02-10:
There is a new user's manual now available on the [wiki](https://github.com/szpajder/RTLSDR-Airband/wiki) - hopefully less confusing than the previous one in the README.

Closing.

closed 2018-02-10 by szpajder