rtl-airband/RTLSDR-Airband #81: raw

mdickw, 2018-02-08:
Does the RAW output work in the latest unstable release? I saw that and got excited. What are the specs? Everything I've tried has been noise or garbled.

szpajder, 2018-02-08:
Docs (beta) are here:
https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-rawfile-outputs

Examples with csdr (https://github.com/simonyiszk/csdr):

AM:

cat file.cs16 | csdr convert_s16_f | csdr amdemod_cf | csdr fastdcblock_ff
| csdr limit_ff | mplayer -cache 1024 -quiet -rawaudio
samplesize=2:channels=1:rate=16000 -demuxer rawaudio -

(if you have NFM support disabled in rtl_airband, then the rate is 8000
instead of 16000.

NFM:

cat file.cs16 | csdr convert_s16_f | csdr fmdemod_quadri_cf | csdr limit_ff
| csdr deemphasis_wfm_ff 16000 100e-6 | csdr convert_f_s16 | mplayer -cache
1024 -quiet -rawaudio samplesize=2:channels=1:rate=16000 -demuxer rawaudio -

or just plug the file into Gnuradio's flowgraph as a file input source and
do whatever you want - scopes, waterfalls, etc.

szpajder, 2018-02-08:
Correction for the AM example:

```
cat file.cs16 | csdr convert_s16_f | csdr amdemod_cf | csdr fastdcblock_ff | csdr limit_ff | csdr convert_f_s16 | mplayer -cache 1024 -quiet -rawaudio samplesize=2:channels=1:rate=16000 -demuxer rawaudio - 
```

Sorry.

szpajder, 2018-02-10:
rawfile support has been released in version 3.0.0.

Closing.

closed 2018-02-10 by szpajder