rtl-airband/RTLSDR-Airband #82: CTCSS filter request

itnavigate, 2018-02-10:
I find myself needing to remove CTCSS  tone from a series of NFM feeds to broadcastify.

Is this something that cane be (easily?) implemented with rtl_airband?

szpajder, 2018-02-10:
Yep, I have a couple of ideas related to audio filtering on my TODO list, including this one.

rec9140, 2018-02-14:
Any chance that this will also allow the code to select only a given PL on channels which have multiple users ie: one channel in my area is shared by a local aeromedevac and due to location and signal another user I don't care about, only difference is PL.  In the setup you would specify the pl=127.3 or something.

szpajder, 2018-02-14:
I might try it, but you'll be testing it :) I don't have a lot of CTCSS signals here around.

itnavigate, 2018-02-15:
I'm happy to test. I have sufficient ctcss

szpajder, 2018-09-19:
Nope. When it is, I'll post info here.

redrob, 2019-06-19:
+1 for CTCSS Support. :)

dh47, 2019-10-06:
Just checking in to see if any progress has been made?

szpajder, 2019-10-11:
Not a whole lot, but there is support for bandpass filtering in outputs which use MP3 compression, so it can be used to filter out CTCSS tones as well. See https://github.com/szpajder/RTLSDR-Airband/pull/112 .

b4sh, 2019-10-17:
My method to filter CTCSS (on client side).

play -t mp3 http://192.168.0.80:8000/radioscanner highpass 300 lowpass 3000

#112 not fork for me.

charlie-foxtrot, 2020-05-31:
Doesn't select only a given PL, but #159 adds a notch filter that can be used to remove a CTCSS tone

charlie-foxtrot, 2023-04-21:
CTCSS support in #368

closed 2023-04-21 by charlie-foxtrot