rtl-airband/RTLSDR-Airband #83: Cannot install latest stable build (3.0.0)

Kaptronik, 2018-02-11:
Hi,

Following your guide, I can't seem to be able to install 3.0.0. Can you please help?

I first downloaded the tarball, uncompressed it. That works.

Now, I cd into the directory, type "make PLATFORM=x86" and I get the following:
`fatal: Not a git repository (or any of the parent directories): .git
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DWITH_RTLSDR   -c -o rtl_airband.o rtl_airband.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DWITH_RTLSDR   -c -o input-common.o input-common.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DWITH_RTLSDR   -c -o input-helpers.o input-helpers.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DWITH_RTLSDR   -c -o output.o output.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DWITH_RTLSDR   -c -o config.o config.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DWITH_RTLSDR   -c -o util.o util.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DWITH_RTLSDR   -c -o mixer.o mixer.cpp
g++ -O3 -g -Wall -DSYSCONFDIR=\"/usr/local/etc\" -DDEBUG=0 -march=native -DWITH_RTLSDR   -c -o input-rtlsdr.o input-rtlsdr.cpp
g++ -rdynamic  rtl_airband.o input-common.o input-helpers.o output.o config.o util.o mixer.o input-rtlsdr.o  -lrt -lm -ldl -lvorbisenc -lmp3lame -lshout -lpthread -lconfig++ -lfftw3f -lrtlsdr -o rtl_airband
`
Any help would be greatly appreciated.

Francis-

starvald, 2018-02-11:
just tried it and got the 'fatal: Not a git repository' error too, however rtl_airband is being built ok.

Kaptronik, 2018-02-11:
Hmmm weird, it seems to run ok indeed. Thank you

closed 2018-02-11 by Kaptronik