rtl-airband/RTLSDR-Airband #84: Squelch bug?

spotter-ssol, 2018-02-13:
I've got this strange behavior regarding manual squelch levels:
[image]
Even after the end of transmission and manual squelch level up to 4x the noise floor, audio keeps streaming indefinitely. This is how it look like in Audacity:
[image]
After some testing I've discovered that this behavior only occurs when compiled without NFM support.

szpajder, 2018-02-13:
Yep, it's broken. Check out the current unstable branch - should be OK now.

spotter-ssol, 2018-02-13:
Solved! Thanks!

spotter-ssol, 2018-08-17:
Today I was testing some configs and got the same behaviour, but now in **scan mode**. Maybe the same bug on another portion of code related to scan?
[image]

szpajder, 2018-08-24:
Do you have any reliable way of reproducing this problem? Does it happen very often?

I've tried several configs, various bands, signal strengths and squelch settings, but it never happened to me so far.

spotter-ssol, 2018-08-29:
I did some more testing and discovered that this happens very often (both multichanel and scan) when I set the squelch 3x (or less) the noise floor level. Audio keeps streaming for about 5 to 15 seconds then usually closes itself. When squelch is set 4x or higher this problem virtually disapears.

WeKnowMC, 2018-08-30:
I have also been experiencing this problem occasionally. It seems like it is most apparent when listening to weak signals, which sometimes dip in and out of the squelch - but it doesn't happen enough for me to reliably confirm this.

szpajder, 2018-08-31:
Please check the current code from unstable branch. I have applied a small tweak, and now it should behave better.

spotter-ssol, 2018-09-01:
Tested with squelch as low as 2x times the noise floor and no sign of problems. Now I can properly track weak signals.
Thank you, sir. I'm having A LOT of fun with this software. Your work is much appreciated!

closed 2018-09-01 by spotter-ssol